Smoothieware/Smoothieware #1303: CartGridStrategy G29 incorrect ysize

CapnBry, 2018-04-05:
On current edge HEAD, the CartGridStrategy's code uses the xsize for ysize when performing a G29 unless overriden by the gcode. There's just a [typo in the assignment I believe](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/zprobe/CartGridStrategy.cpp#L359) where it assigns `_y_size = this->x_size;`

While I'm talking about this G29, I find it inconsistent that when performing the G29, the probe offset is included in the opening bed probe `CartGridStrategy::findBed()` but not in the actual G29 grid probe. My printer starts the G29 no problem but when it enters the loop to start the grid, it moves the nozzle to 0,0 for the first zprobe but now the probe is off the edge and crams the nozzle into the bed. Shouldn't the probe code actually use the probe's position and not the toolhead/nozzle position? In [CartGridStrategy::probe_grid()](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/zprobe/CartGridStrategy.cpp#L340)
```
-            if(!zprobe->doProbeAt(mm, x, y)) return false;
+            if(!zprobe->doProbeAt(mm, x - X_PROBE_OFFSET_FROM_EXTRUDER, y - Y_PROBE_OFFSET_FROM_EXTRUDER)) return false;
```

wolfmanjm, 2018-04-05:
Thank you for finding this. it has now been corrected.

As for the other issue, I find it debatable, as one is a probe the other is a test scan, and I prefer to not include the probe offsets in the test scan at this time, although I will consider it.

closed 2018-04-05 by wolfmanjm

CapnBry, 2018-04-05:
No worries. You're right about the G29, I'm not sure if it should either. I just thought I'd bring it up in case it was an oversight.

Thanks for the quick fix!