Smoothieware/Smoothieware #1309: G28.2 Z does not always move to retract position when finished.

Zaaphod, 2018-04-17:
#### What were you doing?

G28.2 Z to home only Z Axis

#### What did you expect to happen?

Z Axis to home the same as with G28.2 as follows:
   Move at fast home speed to Z limit switch
   move away from limit switch the retract distance
   Move at slow home speed back to Z limit switch
   move away from limit switch the retract distance

#### What happened instead?

Final retract was skipped:
   Move at fast home speed to Z limit switch
   move away from limit switch the retract distance
   Move at slow home speed back to Z limit switch
   Does NOT move away from limit switch the retract distance
   it just stays on the switch with the switch active.

Please note this is in intermittent issue, I can do G28.2 10 times in a row, and 4 out of the 10 times it will perform the retract, the other 6 times it will not... I cannot figure out what causes it to work, or to not work... some days I can't get it to do the retract at all, other days it happens all the time..  I'm baffled by this and don't know where to look.  This happens on all my machines, but for some reason I can't figure out, some machines do it a LOT more often than others. 

#### Branch & Commit or Version of Smoothieware

Current Edge and several previous versions

version
Build version: edge-c819aa6, Build date: Apr  4 2018 16:52:33, MCU: LPC1769, System Clock: 120MHz
  CNC Build 3 axis

#### Hardware model you are using (Make, and version)

Smoothieboard 0X

#### Host and Operating System running the Host

Production Automation Host and 
puTTY terminal

#### Link to and console logs

Log of puTTY session showing repeated G28.2 Z commands followed by ? after OK was received
https://pastebin.com/p4Gzt2PB
    
#### Link to running config and M503 results
I tried both my standard config here:
https://pastebin.com/m9SyHSui

And config from current repository with enough modifications for my machine to work... here:
https://pastebin.com/Ka5Ce8Xe

M503:
https://pastebin.com/HnmPR7n3

wolfmanjm, 2018-04-17:
This is a H/W issue. Usually caused by debounce. Please experiment with different amounts of debounce to fix it.

I do not believe this to be a s/w problem at all.

closed 2018-04-17 by wolfmanjm

Zaaphod, 2018-04-17:
I will try debounce.   I don't really understand how it can be debounce though, the switch is stopping motion at precisely the correct position..  it's just not doing the second retract after it's already finished with the switch..  It ALWAYS does the first retract fine, and the switch always stops motion both times, it's only leaving out the second retract intermittently.    I don't know what's different between the first retract and the second retract that could cause this... but I suspect it's something along the lines of it not moving because the limit switches got turned back on before it was done with the second retract and that's what's preventing it from moving.. as the switch is still active.

MitchelBerberich, 2019-05-04:
Hi. I'm having the same problem.

Maybe I can add some info so you are able to reproduce the bug (I think it is a bug and not a hardware problem): After homing with G28 (I use "G28", not "G28.2") smoothieware does the retract as defined by "beta_homing_retract_mm". (y-axis of a core-xy printer, "corexy_homing=true"). On any subsequent call to G28 no retract happens on the y-axis until the board is resetted. Then retraction is ok again ... exactly one time, then it fails again until the next reset / power-off-on.

Maybe there is an variable which is not properly re(initialized) after or before G-code execution?

Finally I found out, that the problem only occurs when i define "homing_order YXZ". Leaving that commented out, there is no problem with retraction.

If you need more info just ask. I just wanted to report the most important info first, as short as possible. I don't know github bugtracker, so I hope adding this comment will re-open the issue again.

MitchelBerberich, 2019-05-04:
Hi. One more comment / finding:

Looking at the code, I figured out a "better" way for homing which I can use as a workaround and which - one more time - points into the direction of a software problem:

Instead of relying on "alpha/beta_homing_retract_mm -5", I now disable this by "alpha/beta_homing_retract_mm -0" and use "alpha/beta_min -2" in combination with "move_to_origin_after_home true". This approach is working everytime.

I haven't found out yet, why homing_retract fails, but as unrelyable "back_off_home(haxis);" is executed after "move_to_origin(haxis);" (which is working-flawlessly) in https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/endstops/Endstops.cpp [current edge, line 949] this again points to a software problem and not a hardware problem.

Sorry, that I cannot do more at the moment - It's the first time I looked at this project and I have to finish work on my printer first, before spending more time on getting a deeper understanding of smoothieware internals. Hope you (@wolfmanjm) read my comments and reopen the issue again, so that all info is kept in one place. Otherwise I will open a new issue in about a week, so that info is not lost.

wolfmanjm, 2019-05-04:
@MitchelBerberich I am sorry to report but this is NOT a bug. This is exactly how it is supposed to be.

The homing cycle is as follows...

1. Move fast until endstop is hit
2. back off the retract distance
3. move towards endstop slowly until triggered.
4. STAY THERE unless...
  a. limits are enabled, it will now back off the endstop by retract
  b. move_to_origin is set, it will now move to origin/park position

I suggest you do not reopen this as it is not a bug. If you want to add a feature that always retractrs after and endstop is triggered you may submit a PR for it. It would be relatively simple as you would just pretend limits are set.

MitchelBerberich, 2019-05-04:
@wolfmanjm 

But I have alpha/beta/gamma_limit_enable all set to true so according to your step 4.a it shoud do a (second) retract after seeking the endstop with slow speed (3). But that does not happen. To be more precise: Retract happens on the first call to G28 after power-on / reset. First time homing with G28 printer goes to position 5,5. On all subsequent calls to G28, the printer after homing goes to position 5,0.

So at least we have an inconsistent behaviour here (and therefor a software problem) - no matter if retract after homing is correct or not.

wolfmanjm, 2019-05-07:
@MitchelBerberich I have examined the code and I see no where that the retract could be ignored EXCEPT if the endstop is no longer triggered...  https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/endstops/Endstops.cpp#L507

So I stick with the assumption this is a H/W issue, and the check is done so quickly after the endstop is hit that it is still bouncing and happens to look as if it is no longer triggered so the retract is not done. You can check this by either increasing the debounce_cnt (to a big number) or commenting out that check and see if it still exhibits the same behavior. As H/W debounce is unpredictable it is not surprising that this seems inconsistent.