Smoothieware/Smoothieware #1332: soft endstop

mcix, 2018-07-05:
For my machine I have only one endstop on each axis, therefore I was trying to setup my smoothieboard configuration to use soft_endstops. From this feuture: https://github.com/Smoothieware/Smoothieware/pull/1149

When I enable the soft endstop after homing I always get the following error: (returned from using the G0 or G1 command)
> Error: Soft Endstop X was exceeded - entire move ignored

When I look at the following code, it seems that all axises are always checked, is this command always called with all targets? Even though I execute a command like G1 X100?

https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/Robot.cpp#L1171
```
if(soft_endstop_enabled) {
        for (int i = 0; i <= Z_AXIS; ++i) {
            if(!is_homed(i)) continue;
            if( (!isnan(soft_endstop_min[i]) && transformed_target[i] < soft_endstop_min[i]) || (!isnan(soft_endstop_max[i]) && transformed_target[i] > soft_endstop_max[i]) ) {
                if(soft_endstop_halt) {
                    if(THEKERNEL->is_grbl_mode()) {
                        THEKERNEL->streams->printf("error:");
                    }else{
                        THEKERNEL->streams->printf("Error: ");
                    }
                    THEKERNEL->streams->printf("Soft Endstop %c was exceeded - reset or $X or M999 required\n", i+'X');
                    THEKERNEL->call_event(ON_HALT, nullptr);
                    return false;
                //} else if(soft_endstop_truncate) {
                    // TODO VERY hard to do need to go back and change the target, and calculate intercept with the edge
                    // and store all preceding vectors that have on eor more points ourtside of bounds so we can create a propper clip against the boundaries
                } else {
                    // ignore it
                    if(THEKERNEL->is_grbl_mode()) {
                        THEKERNEL->streams->printf("error:");
                    }else{
                        THEKERNEL->streams->printf("Error: ");
                    }
                    THEKERNEL->streams->printf("Soft Endstop %c was exceeded - entire move ignored\n", i+'X');
                    return false;
                }
            }
        }
    }
```

I don't really see any other issues that would cause this issue, if a do a get-config command the x_min, x_max etc.. settings are correct.

Does anyone see any issues here? Or can someone give me some pointers on debugging this issue?

I am using the following configuration:

```
soft_endstop.enable                          true             # enable soft endstops
soft_endstop.halt                            true             # HALT when soft endstop triggered (safest) when set to false the entire move will just be ignored
soft_endstop.x_min                           0                # minimum X
soft_endstop.x_max                           820              # maximum X
soft_endstop.y_min                           0                # minimum Y
soft_endstop.y_max                           380              # maximum Y
#soft_endstop.z_min                          0                # minimum Z NOTE leave commented out to disable checking for this axis
#soft_endstop.z_max                          40               # maximum Z NOTE leave commented out to disable checking for this axis

```

wolfmanjm, 2018-07-05:
this is simply a configuration error on your part. Yes all axis are checked all the time as required.
If you do not want an axis checked then you need to NOT configure that axis.

closed 2018-07-05 by wolfmanjm

wolfmanjm, 2018-07-05:
also you have NOT followed the guidelines for filing an issue. Please read them before posting again.