Smoothieware/Smoothieware #1336: What's the max speed of smoothie & board?

MS1987, 2018-08-09:
I tried to config the "x_axis_max_speed" to be 6000, and it could reach this max speed; 
Afterwards I tried to config the "x_axis_max_speed" to be 100000, and control x motor move by "G1 X10000 F60000", 1000mm/s, and config 80 pulse per mm, so it could generate 80000 pulse per second in theory. However, it only generated about 30k pulse per second on fact. And no matter how I increase the max speed config and the speed para, it could not beyond 30k.
I doubt what is the max speed of this firmware and board, as I have heard that it can reach 100k pulse per second.

arthurwolf, 2018-08-09:
what does the "version" command answer ?

wolfmanjm, 2018-08-09:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and **modify your issue** accordingly. **(DO NOT CREATE A NEW ISSUE**)

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2018-08-09:
FWIW It has been confirmed that it can do 100KHz step pulses with a logic probe. You probably have not configured it properly or you are not really understanding how things work.

closed 2018-08-09 by wolfmanjm

MS1987, 2018-08-09:
Find the result is , my firmware version is too old. The current version is ok.