Smoothieware/Smoothieware #1347: M110 command is not working as described in Documentation

luc-github, 2018-09-26:
According https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/communication/GcodeDispatch.cpp#L86-L99

To reset line number seems need to type :
```
N{New Line Position} M110
```
but documentation http://smoothieware.org/supported-g-codes give different syntax
```
M110 N{New Line Position}
```
[image]

wolfmanjm, 2018-09-26:
it seems to work fine.

closed 2018-09-26 by wolfmanjm

luc-github, 2018-09-26:
you mean M110 N{New Line Position}  work fine ?
It is not on my side

wolfmanjm, 2018-09-26:
As far as I know ALL hosts send Nxxx M110 to reset the line number. This seems to be the defacto standard for reprap.
It would be easy enough for you to add code to allow M110 Nxxx if you insist on going against the defacto standard.
I'd also check to see what the NIST standard says about M110 (if it says anything as I suspect this is a reprap addition).