Smoothieware/Smoothieware #1353: M907 does not set currents for expected axes, or docs are missing

docdawning, 2018-12-07:
#### What were you doing?
I've had some filament grinding issues and the odd axis getting stuck during travel. I also have noticed one of my axes stepper ICs gets rather hot, while the others seem reasonable.

#### What did you expect to happen?
I expected my axes to work consistently well, or poorly. And my extruder's grinding has been confusing given I've got a nice heafty E3D Titan and I'm not pushing it particularly hard.

#### What happened instead?
... This template doesn't really fit. M907 doesn't behave as I'd expect based on the docs on the wiki.

#### Branch & Commit or Version of Smoothieware
FIRMWARE_VERSION:edge-a06bb6a, X-FIRMWARE_BUILD_DATE:Nov 30 2018 20:34:40, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-GRBL_MODE:0, X-CNC:0, X-MSD:1

#### Hardware model you are using (Make, and version)
Smoothieboard 5X v1.0a

#### Host and Operating System running the Host
Octoprint, on a Raspberry Pi (OctoPi)

#### Link to and console logs
This isn't really going to help you... But here you go..
https://gist.github.com/docdawning/06df7b3a28e6a3d80baaa0d39604475d
    
#### Link to running config and M503 results
**M115 Output**
FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-a06bb6a, X-FIRMWARE_BUILD_DATE:Nov 30 2018 20:34:40, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-GRBL_MODE:0, X-CNC:0, X-MSD:1

**config File**
[Smoothieware_config.txt](https://github.com/Smoothieware/Smoothieware/files/2656497/Smoothieware_config.txt)

**config-override File**
[Smoothieware_config-override.txt](https://github.com/Smoothieware/Smoothieware/files/2656498/Smoothieware_config-override.txt)

**Explanation**
[The docs](http://smoothieware.org/supported-g-codes) explain M907 as:
M907 | Set Current control via digipot for each axis (current in amps) | M907 X1.0 Y1.0 Z1.0 A1.5

For a Delta machine, I interpret this, to mean the A axis is the first extruder. However, very obviously, when I enter M907 with small values for everything but the A axis, my extruder is rather quiet and fails to push filament. If I change only the X axis to have a large current, then my extruder works great and sounds super loud. Either I feel the docs could use a little more detail, or this is bug.

**Video of unexpected behaviour**
https://youtu.be/mW6ly3mEzUc

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2018-12-07:
first your config-override is incorrect as there is no M907 entry, this may mean your current control module is not enabled and/or your config is corrupted. I suggest you try a stock config from the github and provide the log results of M503.
The S/W has been working for several years and there are no known bugs in the S/W.
If after fixing your config and config-override (or delete the config-override) you still get the issue with X setting A then there must be a H/W bug on your board. You can check if it is working by testing the the Vref for each driver.

closed 2018-12-07 by wolfmanjm

wolfmanjm, 2018-12-07:
The log also shows...
M503...
Recv: ;Digipot Motor currents:
Recv: M907 X0.30000 Y0.30000 Z0.30000 A1.45000

So your A (E) is set correctly.

However there are many M907 in the gcode file this is illegal, as M907 can only be set once and is not recommended to be set more than once after boot. If you have M907 in the gcode file then You must remove them. Many of the M907 in the gcode file sets the extruder current to 0.3. This is obviously user error.

As for the documentation this is a user wiki so feel free to add documentation if you see fit.

docdawning, 2018-12-08:
@wolfmanjm If it's a predictable user error to set M907 more than the desired number of times, maybe the firmware could behave that way. 

1. Only accept M907 once before the first Motor-On event.
2. Subsequent M907's return a message: "Nope, not doing this"
3. Create a new M-code for actually forcing it, for edge-cases who want this

wolfmanjm, 2018-12-08:
this has turned out to not be a firmware issue current mappings are to the driver not to the axis. this is shown in the wiki.

docdawning, 2018-12-12:
For future ref, I've changed the config template file so the current parameters in the config file are no longer commented as for X, Y, Z, Delta, Epsilon axes, but rather their corresponding M1, M2, .. ,M5 driver.