Smoothieware/Smoothieware #1356: $G does not return an "ok", which breaks bCNC

vatral, 2018-12-15:
#### What were you doing?

I'm using a CNC-converted Proxxon MF-70. I'm using bCNC (latest from git) for the control.

#### What did you expect to happen?

A job should be performed to completion.

#### What happened instead?

Often, the jobs randomly stall somewhere in the middle. bCNC accumulates commands in the buffer. What appears to happen is that confirmations get out of sync, and bCNC waits for confirmations that won't arrive.

Investigating the issue, I found that $G specifically is a problem: when used, bCNC treats it as unconfirmed. During a job, confirmations for subsequent commands get treated as a confirmation for $H, and random commands end up in the buffer.

The result is that if there's a single $G that's executed while bCNC runs, bCNC will either get stuck somewhere in the middle of a job, or get to the end and still think there's something left to do.

I have investigated grbl. It always confirms a $G:

https://github.com/grbl/grbl/blob/master/grbl/protocol.c#L49

Then I investigated Smoothieware. Turns out this behavior is intentional and was added recently, in commit 54f0098a4

Unfortunately it causes exactly the opposite effect from what's intended.

#### Branch & Commit or Version of Smoothieware

FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-a06bb6a, X-FIRMWARE_BUILD_DATE:Nov 30 2018 20:34:58, X-SYSTEM_CLOCK:120MHz, X-AXES:3, X-GRBL_MODE:1, X-CNC:1, X-MSD:1

#### Hardware model you are using (Make, and version)

Smoothieboard 5X V1.1

#### Host and Operating System running the Host

Linux, bCNC from git

#### Link to and console logs

Here is a full log of a failed bCNC job:

https://github.com/vlachoudis/bCNC/files/2669681/bCNC.log

In the middle of the job, bCNC stops submitting commands even though the job is not nearly done. $G seems to be at the very least a contributing factor, if not the entire problem.
    
#### Link to running config and M503 results

[config.txt](https://github.com/Smoothieware/Smoothieware/files/2683057/config.txt)

https://pastebin.com/sCpXS460

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2018-12-15:
This is true. I did this to fix the issue in smoopi.

I have two suggestions.
1. build you own version of smoothie and uncomment that line.
2. Fix bcnc for the smoothie flavor where $G is an instant command that does not return ok.

I will consider if I should re add this though as I think I fixed the issue in Smoopi.

bCNC has many issues with smoothie though and although I have reported many of them it seems to have been decided that smoothie is not an important enough platform for bCNC to work correctly with, and it is not possible to change smoothie enough to be 100% grbl compatible.

closed 2018-12-15 by wolfmanjm

vatral, 2018-12-15:
But if you wrote Smoopi, why not change Smoopi instead? It seems like an odd deviation from the general pattern of everything being confirmed, and it probably breaks other programs as well. 

Also, bCNC is the very first thing on http://smoothieware.org/CNC-Mill-guide#cnc while Smoopi isn't even mentioned.

Based on that, I assumed that it's supposed to work.

If bCNC has known issues, and Smoopi is preferred, I think the documentation should be updated to reflect that.

wolfmanjm, 2018-12-15:
Well it was not easily fixed in Smoopi, and at the time I didn't think it would break bcnc.
As I said I'll consider reversing this change, in the meantime you can easily just uncomment that line to send ok on $G.

wolfmanjm, 2018-12-15:
Not everything is confirmed even in grbl. this is part of the protocol that is not documented and is implemented differently in most firmwares.
I think I'll just create a different command for smoopi, and put $G back the way it was.

vatral, 2018-12-15:
I see, fair enough. I'm new at this and don't really know what's standard and what not.

Now that you mentioned Smoopi though I want to give it a try, but the situation of needing to change firmware back and forth to get one or the other working is a bit troublesome, so I'm considering trying to make bCNC work around it.

Can you suggest a way to find out which behavior would be needed? Smoothieware doesn't seem to have an actual version number, and git revisions are not sequential.

Thanks!

wolfmanjm, 2018-12-15:
#1357 fixes this.

wolfmanjm, 2018-12-15:
For the most part Smoopi will work with only the latest builds. It is designed to work with both 3d printing and CNC which is quite hard.