Smoothieware/Smoothieware #1361: problem delta grid compensation

vivpopo, 2019-01-06:
Hi ,

Since a copple of months ago i don't upgrade the last smoothie FW because the delta grid compensation didn't work.
It works with the release of 30/12/2017.
After this release I never achieved  runing the grid probing.
I searched on the web but found until know the solution .
I use a TFT touch screen and disabled the leveling in the config tft file . KO
Tried to chenge Z probe initial high at 20 mm and into the delta grid compensation high KO

My files With M503 and M115 information

[M115_M503.txt](https://github.com/Smoothieware/Smoothieware/files/2730948/M115_M503.txt)

My config file : 

[config.txt](https://github.com/Smoothieware/Smoothieware/files/2730951/config.txt)

Below the starting script  I use to probe the bed :

G28 X Y 		; Home X and Y
G1 X0 Y0 F6000	; Go to center of bed
M106		;Turn ON fan
M280 S10.6		;Erase probe error
M280 S3.0 		; Deploy BLT probe pin
G31 		; Probe the bed and turn on compensation
G1 X0 Y0 F6000	; Go to center of bed 
G30 Z1.3		; Z Probe Offset
M280 S7.0 		; Retract probe pin
G92 E0 		; 0 Extrution distance
G1 Z0.25 F500 	; Z height 0.25
G1 X-106.0 F2000 	; Move to prime
G1 Y-80.0 E5.0 F1000.0 	; Prime
G1 Y50.0 E8.0 F1000.0 	; Prime
G92 E0 		; 0 Extrution distance

Before the initial probe of the bed ran well . 

Now the bl touch still stay on the center of the bed and cancel the bed probing , and follow the print .

Before  I was probing for each print .Now I registered the grid which was been made with the old smoothie FW.
I would like to make a new probing bed but mith the last FW I can't . It works with the old smoothie FW .

As I activated the error message on my tft and it gave me this error  (sorry I haven't got the whole message because screen size : 

First line on the screen  : Error : is_square has been removed , please 
Second line on the screen :  e the new rectangular_grid strategy inst(ead)
third  line on the screen : confirm

Thx in advance for any help

arthurwolf, 2019-01-06:
Your config file is updated. Please refer to the documentation, it says you
need to update your config file to the latest example config file if
updating your firmware. Get the config file from the wiki and port your
values one by one to the new format. Also make sure you read the upgrade
notes ( again this is in the documentation ).

What board type is this ?

What is the result of the "version" command ?

arthurwolf, 2019-01-06:
I meant "outdated", not "updated".

wolfmanjm, 2019-01-06:
This is not a bug, so please do not use issues for help.

Delta grid compensation has been workign for years now and I use it daily. so the issue is your config or simply not using it correctly.

Please read the wiki for the docs.

Also DO NOT run G32 from a gcode file it is NOT designed to be run automatically, it is to be run manually and the results saved.

closed 2019-01-06 by wolfmanjm

vivpopo, 2019-01-07:
Ok thanks for quick answers !
Will give a try with a new config file .
Thanks !

vivpopo, 2019-01-07:
Hi All ,
Many for your posts I noticed the mistake I did . I found into the deltagrid example file what I didn't change to make it working !
Sorry for disturbing you and without you I would be maybe until into searching what was wrong !! 
I now understood the error message on my lcd panel which pointed me to my error.
so Sorry but Many thanks ! Solved