Smoothieware/Smoothieware #1362: Print failure always in line N16777217

GitOCM, 2019-01-22:
Hello,

when printing large files (for example 18.000.000 lines of gcode) the printer stops executing the file in line N16777217 (what is exactly 2^24 + 1)

When the printer reaches line N16777217 following communication is repeating again and again in the terminal:
`Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
| Send: N16777217 G1 X-42.274 Y30.520 E0.0033*72
| Recv: rs N16777217
.
.
.`

The Smoothieboard is connected via USB to octoprint running on a Pi 3 B.

Here the results of M115 and M503:
`Send: M115
Recv: FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:~11myvar2, X-FIRMWARE_BUILD_DATE:Oct 21 2018 13:37:14, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-GRBL_MODE:0, X-CNC:0, X-MSD:1

Send: M503
Recv: ; No config override
Recv: ;Steps per unit:
Recv: M92 X80.00000 Y80.00000 Z33109.00000 A127.60000 B14400.00000
Recv: ;Acceleration mm/sec^2:
Recv: M204 S3000.00000
Recv: ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
Recv: M205 X0.05000 Z-1.00000 S0.00000
Recv: ;Max cartesian feedrates in mm/sec:
Recv: M203 X1000.00000 Y1000.00000 Z5.00000 S-1.00000
Recv: ;Max actuator feedrates in mm/sec:
Recv: M203.1 X500.00000 Y500.00000 Z3.00000 A500.00000 B6.00000
Recv: ;Digipot Motor currents:
Recv: M907 X1.50000 Y1.50000 Z1.50000 A2.00000 B1.50000
Recv: ;Home offset (mm):
Recv: M206 X0.00 Y0.00 Z0.00 A0.00 B0.00
Recv: 
Recv: ok`

A solution is needed to print files with more lines than 16777217

Many thanks!

wolfmanjm, 2019-01-22:
I quick look I see that a fix could be easy..

change these lines 
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/communication/GcodeDispatch.cpp#L89

to...

```
            ln = (int) full_line.get_int('N');
            int chksum = (int) full_line.get_int('*');
```

Let me know if that fixes it. Thanks.

GitOCM, 2019-01-29:
I modified the code and now it is possible to execute more than 2^24 lines of gcode. I tested with around 18.000.000 lines of gcode.

Many thanks for your help.

Can you tell me what is the new limit with the integer implementation now?

wolfmanjm, 2019-01-29:
2^31

PR #1363 (2019-01-29): fix line number range
Fixes #1362
closed 2019-01-29 by wolfmanjm