Smoothieware/Smoothieware #1366: Board stops moving after rectangular grid levelling

FrenkK, 2019-02-13:
#### What were you doing?
I turn on the board and home. It works normally.
I do a rectangular grid levelling (G32 X96 Y10 A130 B70 I7 J5), it completes normally.
Here the problem starts - any movement command after finished levelling causes a short high-pitched whine in the motors once and then it becomes impossible to make any move using either G0 or G1 or probing using G30. Homing will work, but movement commands still doesn't work after that.
This happens both with latest CNC and normal CNC build.

#### What did you expect to happen?
Normal move commands should work after rectangular grid levelling.

#### What happened instead?
Move commands stop working properly. No "OK" after a movement command is entered.

#### Branch & Commit or Version of Smoothieware
**M115:**
FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-a06bb6a, X-FIRMWARE_BUILD_DATE:Nov 30 2018 20:34:58, X-SYSTEM_CLOCK:120MHz, X-AXES:3, X-GRBL_MODE:1, X-CNC:1, X-MSD:1

Happens with latest CNC too.

#### Hardware model you are using (Make, and version)
Original 4XC Smoothieboard v1.1

#### Host and Operating System running the Host
Pronterface on Win 10 x64 or CNCjs on Raspberry Pi 3

#### Link to running config and M503 results
M503: https://pastebin.com/jkL5mZJh
Config: https://pastebin.com/CDvHJBQw

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Copy-paste this template **completely**. Do not skip any lines!

It might happen that you are asked to provide a more thorough crash log which you can find out how to do [here](http://smoothieware.org/mri-debugging
)

wolfmanjm, 2019-02-13:
you need to reduce your feed rate. this probably means the z axis is having to move too fast to compensate. There is no bug in smoothie with this function. try setting your max z axis feedrate much lower.

closed 2019-02-13 by wolfmanjm

FrenkK, 2019-02-13:
I'm sorry, but have you read my description? It works normally (movement commands, probing, everything) until I do the levelling, then it goes crazy.

wolfmanjm, 2019-02-13:
Yes I read it, and I told you what the issue is. When you level then every movement causes the Z to move up and down to compensate, unless you set your max Z rate properly it can cause the Z to try to move faster than it is capable, even if you are not actually moving in Z. That is what levelling does.

wolfmanjm, 2019-02-13:
If you disable the grid levelling does it start moving again ok?

FrenkK, 2019-02-13:
No, this can't be the issue.
My machine is symmetrical and all three axes can do 2000 mm/s. I have the speeds capped at 1500 mm/s for safety. The problem appears even with probing, which uses 5 mm/s speed in Z-direction only, way below any possible skipping issues.
I'll check what happens if the levelling is disabled, but not only movement stops working, the spindle commands are ignored too.

FrenkK, 2019-02-13:
I have a possible answer: Z-probe parameters in config are in mm/sec, while the rest of the speeds for config are in mm/min. I'll check if changing these makes a difference.

FrenkK, 2019-02-13:
Ok, first of all, I went through all the config settings and changed all the mm/sec values to reasonable ones (never more than 25, what I know is safe). It is confusing that both units are used.

I now tried levelling again, but the problem remains. I also checked and after I send M370 to disable levelling, the first move (exclusively Z-axis) is still too fast, causing skipping. After that, it starts working normally.

So, for some reason, the speeds seem to go up after leveling, beyond the limits in config file and what the steppers can handle. After leveling disable, the problem remains for a little while and then clears.

wolfmanjm, 2019-02-13:
you do need go home after turning off the levelling as it could be left in a bad state otherwise.