Smoothieware/Smoothieware #1369: LaserGRBL with Smoothie - Work in Progress

arkypita, 2019-02-26:
Hi @arthurwolf and hi all, I am the developer of LaserGRBL software.
First of all my congratulations for having created a control board so professional and complete. As promised I will try to test it at best and I will try to make the necessary changes in LaserGRBL to make it compatible.

Here I will post all my question/doubt about this task. Here are some of them:

- Smoothie seem not reply at ? if sent as "[immediate command](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-control-commands)" as in grbl scheme, but seem to need a "\n" postfix. Am I right? There are something conceptually like real time command in smoothie or not?
- LaserGRBL use buffered ([character counting](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-protocol-character-counting-recommended-with-reservation)) streaming tecnique. This provides a buffer of 128 in grbl. What is the buffer size for smoothie?
- I see that sometime smoothie send something like grbl welcome message (sending "Smoothie") but i cannot understand exactly when, and why no version info in this message. LaserGRBL actually use welcome message to understand to which board is talking with, and firmware version, to self-configure with firmware capability. Can someone tell me how to be sure that LaserGRBL will receive a welcome message when connect (if possible)?

arthurwolf, 2019-02-26:
Hi !

Thanks for coming over here and looking into this !
Jim Morris ( wolfmanjm ) is our smoothie grbl_mode specialist, he
implemented all of that, and has intimate knowledge of that part of the
interface ( "?" and friends ) as well as the best method to stream to
Smoothie ( which I think the *ideal* method to stream may be different, but
it's possible the "normal grbl" method might still work with Smoothie ? ).
I think I might already be saying things I don't know enough about so I'll
let Jim handle this :)

wolfmanjm, 2019-02-26:
Make sure you are using the latest version of smoothie, and that it is the cnc build (make CNC=1).

? returns immediately there is no OK and it does not require a \n, It sounds like maybe the way you send it is in question?

Smoothie does not use a uart it uses USB CDC, as such there is NO buffering, you MUST NOT do character counting this will not work with smoothie and in fact will cause deadlock within the USB driver. You MUST use a strict ping-pong protocol.. send command wait for ok, send next command etc. This maybe why your ? is not working as expected.

Welcome messages are not supported and cannot be handled reliably due to the way CDC works. So you should send M115 once you are connected then you will get all the info you want.

If you need more help I suggest you look at my Smoopi project, especially the comms.py code which talks reliably to smoothie. https://github.com/wolfmanjm/kivy-smoothie-host

It is useful to note that USB CDC drivers on different platforms works slightly differently as the drivers are different, Linux works very well, windows has some quirks like delays before sending etc. Not sure about MacOS but I think it works more like Linux. The USB CDC does not need resends or line number and checksums as the USB packets are sent with checksums in the USB protocol, and has its own flow control, it is this low level flow control that can cause deadlock if you do not follow strict ping pong protocol, or do not read the return before sending the next, this is not a Smoothie issue it is the way USB CDC works. If you decide to not wait for ok before sending the next command then the USB flow control will kick in, and unless yo are reading the responses in a separate thread you will deadlock. (I have a special fast stream mode in Smoopi which makes use of this allowing USB flow control to be used).

closed 2019-02-26 by wolfmanjm

arthurwolf, 2019-02-26:
This link might be helpful : http://smoothieware.org/grbl-mode

wolfmanjm, 2019-02-27:
BTW if you have further questions continue to ask here.

NOTE that the data returned by the ? query is different to what GRBL (new and old) return. The formatting is the same, but the actual data returned is different.
grbl_mode does not mean we are compatible with grbl, it just changes some gcodes to be more NIST compliant, so really should be called cnc_mode. As GRBL is completely different to smoothie it would not be possible to be 100% compatible.

arkypita, 2019-02-27:
Thanks both; the thing that I love about the open community is the willingness to help one another

>Smoothie does not use a uart it uses USB CDC, as such there is NO buffering, you MUST NOT do character counting [...] use a strict ping-pong protocol.. send command wait for ok, send next command etc.

Fantastic! No problem for me: LaserGRBL support both modes.

>? returns immediately there is no OK and it does not require a \n, It sounds like maybe the way you send it is in question? [...]  This maybe why your ? is not working as expected.

The issue I see is that when I connect to COM port, then I send the single character "?" (byte 0x3F) without any \n my smoothie does not reply at all.

If I send something else (like a gcode or a ?\n) it reply with the string "Smoothie" and from now ahead it reply also to "?". This seems very strange but it's a test I did quickly this morning, so I want to go deeper into it before giving it for sure

> Welcome messages are not supported and cannot be handled reliably due to the way CDC works. So you should send M115 once you are connected then you will get all the info you want.

Ok, thanks.

> the data returned by the ? query is different to what GRBL (new and old) return. The formatting is the same, but the actual data returned is different.

Yes, I already seen some difference (i.e. F: in smoothie transport two numbers, only one in grbl). I am going to manage a compatible parser for both.

wolfmanjm, 2019-02-27:
When you first connect you should send a \n then consume any lines you get, there maybe one or more ok sent due to driver noise etc. Then you can send M115, then after that sending a single ? will return the query string. If it does not then for some reason your host is not transmitting the single character. I did find that some drivers only send lines if not configured correctly.
NOTE that any blank lines sent to Smoothie will return ok and all G/M codes will return ok. Most lowercase commands will also return ok (but some do not). The reason you got ok when you sent ?\n is that smoothie sees the ? responds to that then sees a blank line and sends ok.

arkypita, 2019-03-17:
Hi @arthurwolf, hi Smoothie community
v3.0.10 is now the first LaserGRBL release with experimental smoothie support. 

Smoothie mode can be enabled by selecting smoothie firmware under "grbl-settings" menu.

[image]

I have not tested it with a real engraver, only with the board itself: no motor, no laser connected.
Tested with USB and telnet via ethernet.

Please give some feedback :-)

wolfmanjm, 2019-03-17:
Thank you. As I do not use Windows I can't test it, but do let me know if there are any issues I can help with.

arthurwolf, 2019-03-18:
Shared on social media, hopefully, you get lots of testers soon. Thanks a
lot for the effort !