Smoothieware/Smoothieware #1378: M600 stops heaters even with leave_heaters_on_suspend true

ZhuDaHai, 2019-03-10:
I have M600 in my gcode file to Pause the print. However, the heaters turn off even if 'leave_heaters_on_suspend                     true' is in the config.txt file.

gcode file:
`;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
M117 M600: Filament change pause
G28 X0 Y0           ; Home X and Y
M400                ; Wait for command buffer to clear
M600                ; Ignores all parameters
G28 X0 Y0           ; Home X and Y 
G92 E0              ; Reset Extruder Origin
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;`

config.txt file:
`# Misc Settings
leave_heaters_on_suspend                     true             # For M600 Suspend
`
M115 Output:
`Send: M115
Recv: FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-c819aa6, X-FIRMWARE_BUILD_DATE:Apr  4 2018 16:52:08, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-CNC:0, X-MSD:1`

M503 output:
`Send: M503
Recv: ; config override present: /sd/config-override
Recv: ;Steps per unit:
Recv: M92 X80.00000 Y80.00000 Z400.00000
Recv: ;Acceleration mm/sec^2:
Recv: M204 S3000.00000
Recv: ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
Recv: M205 X0.05000 Z-1.00000 S0.00000
Recv: ;Max cartesian feedrates in mm/sec:
Recv: M203 X500.00000 Y500.00000 Z500.00000
Recv: ;Max actuator feedrates in mm/sec:
Recv: M203.1 X500.00000 Y500.00000 Z250.00000
Recv: ;WCS settings
Recv: G54
Recv: ;Digipot Motor currents:
Recv: M907 X1.00000 Y1.00000 Z1.00000 A1.00000 B1.00000
Recv: ;Home offset (mm):
Recv: M206 X0.00 Y0.00 Z0.00
Recv: ;E Steps per mm:
Recv: M92 E450.0000 P57988
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P57988
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P57988
Recv: ;E acceleration mm/sec��:
Recv: M204 E500.0000 P57988
Recv: ;E max feed rate mm/sec:
Recv: M203 E50.0000 P57988
Recv: ;PID settings:
Recv: M301 S0 P14.4080 I0.6244 D83.1163 X128.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S0 P300.0000
Recv: ;PID settings:
Recv: M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S1 P300.0000
Recv: 
Recv: ok`

I have M601 set in my Custom Menu to resume the print. Which does reheat everything and then resume as expected. But it was my understanding that 'leave_heaters_on_suspend                     true ' would keep the heaters on so they would not have to be re-heated.

`custom_menu.m601.enable                 true
custom_menu.m601.name                   M601_Resume
custom_menu.m601.command                M601
`
This in an Panucatt Azteeg X5 GT mainboard.

thank you for any help with this issue/bug(?)

wolfmanjm, 2019-03-10:
I suspect you have a corrupt config, as you did not submit the entire config as required by the How to Submit an issue I cannot tell. You also did not submit a complete log of the process.

This option is known to work, so that is the only thing I can think of, either that or you have a turn off heaters somewhere in the gcode.

ZhuDaHai, 2019-03-10:
Thanks for the quick response. Here is the complete config.txt file:

```
# Robot module configurations : general handling of movement G-codes and slicing into moves
default_feed_rate                            4000             # Default rate ( mm/minute ) for G1/G2/G3 moves
default_seek_rate                            4000             # Default rate ( mm/minute ) for G0 moves
mm_per_arc_segment                           0.5              # Arcs are cut into segments ( lines ), this is the length for these segments.  Smaller values mean more resolution, higher values mean faster computation
#mm_per_line_segment                          5                # Lines can be cut into segments ( not usefull with cartesian coordinates robots ).

# Arm solution configuration : CoreXY robot. Translates mm positions into stepper positions
arm_solution                                 corexy           # DaHai
alpha_steps_per_mm                           80               # Steps per mm for alpha stepper
beta_steps_per_mm                            80               # Steps per mm for beta stepper
gamma_steps_per_mm                           400              # Steps per mm for gamma stepper 1637.7953

# Planner module configuration : Look-ahead and acceleration configuration
planner_queue_size                           32               # DO NOT CHANGE THIS UNLESS YOU KNOW EXACTLY WHAT YOUR ARE DOING
acceleration                                 3000             # Acceleration in mm/second/second.
#z_acceleration                              500              # Acceleration for Z only moves in mm/s^2, 0 disables it, disabled by default. DO NOT SET ON A DELTA
acceleration_ticks_per_second                1000             # Number of times per second the speed is updated
junction_deviation                           0.05             # Similar to the old "max_jerk", in millimeters, see : https://github.com/grbl/grbl/blob/master/planner.c#L409
                                                              # and https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8 . Lower values mean being more careful, higher values means being faster and have more jerk

# Stepper module configuration
microseconds_per_step_pulse                  1                # Duration of step pulses to stepper drivers, in microseconds
base_stepping_frequency                      100000           # Base frequency for stepping
# Stepper module pins ( ports, and pin numbers, appending "!" to the number will invert a pin )
alpha_step_pin                               2.1              # Pin for alpha stepper step signal
alpha_dir_pin                                0.11!            # Pin for alpha stepper direction
alpha_en_pin                                 0.10             # Pin for alpha enable pin 0.10
alpha_current                                1.0              # X stepper current
x_axis_max_speed                             30000            # mm/min
alpha_max_rate                               30000.0          # mm/min actuator max speed

beta_step_pin                                2.2              # Pin for beta stepper step signal
beta_dir_pin                                 0.20!            # Pin for beta stepper direction
beta_en_pin                                  0.19             # Pin for beta enable
beta_current                                 1.0              # Y stepper current
y_axis_max_speed                             30000            # mm/min
beta_max_rate                                30000.0          # mm/min actuator max speed

gamma_step_pin                               2.3              # Pin for gamma stepper step signal
gamma_dir_pin                                0.22             # Pin for gamma stepper direction
gamma_en_pin                                 0.21             # Pin for gamma enable
gamma_current                                1.0              # Z stepper current
z_axis_max_speed                             30000              # mm/min
gamma_max_rate                               30000.0            # mm/min actuator max speed

# Serial communications configuration ( baud rate default to 9600 if undefined )
uart0.baud_rate                              115200           # Baud rate for the default hardware serial port
second_usb_serial_enable                     false            # This enables a second usb serial port (to have both pronterface and a terminal connected)
#msd_disable                                 false            # disable the MSD (USB SDCARD) when set to true

# Extruder module configuration
extruder.hotend.enable                          true             # Whether to activate the extruder module at all. All configuration is ignored if false
extruder.hotend.steps_per_mm                    450              # Steps per mm for extruder stepper
extruder.hotend.default_feed_rate               600              # Default rate ( mm/minute ) for moves where only the extruder moves
extruder.hotend.acceleration                    500              # Acceleration for the stepper motor mm/sec▒
extruder.hotend.max_speed                       50               # mm/s

extruder.hotend.step_pin                        2.0              # Pin for extruder step signal
extruder.hotend.dir_pin                         0.5!             # Pin for extruder dir signal
extruder.hotend.en_pin                          0.4             # Pin for extruder enable signal

# extruder offset
[... cut ...]
2) The print runs from the SD Card
3) When layer 26 is reached, X and Y are homed
4) Then M600 is reached and the print pauses 
5) The heaters turn OFF
6) Select "M601 Resume" from the Custom Menu
7) The heaters re-heat
8) The print resumes

I expect #5 to Not happen if 'leave_heaters_on_suspend true' is present in the config.txt file as I have it inserted in mine.

Thanks.

Attached is the gCode for this print

[M600 Change-Hevo.zip](https://github.com/Smoothieware/Smoothieware/files/2949446/M600.Change-Hevo.zip)

wolfmanjm, 2019-03-10:
Still no console log, i cannot help you without more information.
What host are you using to send the gcode?
I also think the config maybe too big. Try moving the line that tells it to leave heaters on to the start.

wolfmanjm, 2019-03-10:
Also try using a more recent firmware.

ZhuDaHai, 2019-03-10:
I'm printing from SD Card
I'll try moving the leave heaters line to the top.
What's the size limit of the config.txt file?
I'll try to figure out how to get the consol log.
And if all that fails, I'll upgrade the firmware.
Thanks!

wolfmanjm, 2019-03-10:
also try M600.1 instead of M600
As per the rules you MUST use latest firmware before reporting an issue.

ZhuDaHai, 2019-03-10:
Sorry. I thought it was latest.
Where is M600.1 documented?
The only things I've found in the Smoothieware web site is M600

ZhuDaHai, 2019-03-10:
Ok. Moved the 'leave_heaters_on_suspend true' to just before the Custom Menus and... it works!!!
Thanks!
Issue closed/sort of/ - would be nice to know what the file size limit is of the config.txt file
Cheers!

closed 2019-03-10 by ZhuDaHai