Smoothieware/Smoothieware #1379: CartGridStrategy::doCompensation() does not work for coordinates outside of the probed area

mdelanno, 2019-03-11:
The documentation states that:

> the very latest firmware will not jump when outside the probed area, it will just use the nearest edge heights instead.

But when I'm looking to the code, I see that in this case, the computed indices to access the grid values are wrong. If we have a 7x7 grid, this code:

  int floor_y = (int) Floor(grid_y);

will put 6 (the max) in `floor_y`.

After that, the following code is used to read the point on the right:

  float z4 = grid[floor_x + 1 + (floor_y + 1) * current_grid_x_size];

The problem is that one unit is added to a coordinate which is already to the max : `floor_y + 1`

The computed index will be above the bounds of the array, making the code fails (I don't know how exactly but when I'm testing this, my printer is going crazy).

The firmware should halt the printer in this case because there is no way it can compute the compensation. Even extrapolation can produce results which are not good because the bed is not supposed to be flat and the user will experiment a strange behavior.

wolfmanjm, 2019-03-11:
I don't think you are taking into consideration the previous lines of code, which clamps the area correctly.

```
    // find min/maxes, and handle the case where size is negative (assuming this is possible? Legacy code supported this)
    float min_x = std::min(this->x_start, this->x_start + this->x_size);
    float max_x = std::max(this->x_start, this->x_start + this->x_size);
    float min_y = std::min(this->y_start, this->y_start + this->y_size);
    float max_y = std::max(this->y_start, this->y_start + this->y_size);

    // clamp the input to the bounds of the compensation grid
    // if a point is beyond the bounds of the grid, it will get the offset of the closest grid point
    float x_target = std::min(std::max(target[X_AXIS], min_x), max_x);
    float y_target = std::min(std::max(target[Y_AXIS], min_y), max_y);

    float grid_x = std::max(0.001F, (x_target - this->x_start) / (this->x_size / (this->current_grid_x_size - 1)));
    float grid_y = std::max(0.001F, (y_target - this->y_start) / (this->y_size / (this->current_grid_y_size - 1)));

```

A quick scan of the code looks correct to me.

As you do not provide any information as required by the rules of submitting an Issue I cannot guess  
 what issue you may actually be running into.
However if you do find a bug please submit a pull request for the fix.

wolfmanjm, 2019-03-11:
Also your grid is 7x7, and the floor_y is 6, 6+1 == 7, thus it is NOT out of bounds.
Also if your X size is 100, and you  got to 120, then the target is set to 100, which would again yield 6 as the floor, and 6+1 is 7.
I am not sure what your point is here but I see no bug.

wolfmanjm, 2019-03-11:
Ok maybe I spoke too soon. I'll need to look into this further.
However halting is not an option.

wolfmanjm, 2019-03-11:
This appears to be the solution...
```
float grid_x = std::max(0.001F, std::min(this->current_grid_x_size - 1.001F, (x_target - this->x_start) / (this->x_size / (this->current_grid_x_size - 1))));
float grid_y = std::max(0.001F, std::min(this->current_grid_x_size - 1.001F, (y_target - this->y_start) / (this->y_size / (this->current_grid_y_size - 1))));
```

What do you think? This means that floor_x and floor_y are never more than 5. Thus not going out of bounds.
This is taken pretty much from the deltagrid which does work.

wolfmanjm, 2019-03-11:
I am glad you bought this to my attention. It turns out it is not an issue of being out of the probed area, it looks to me like this never worked for the right and top edges of the probed area. On the right edges it would be getting the level for the left edge on the next line down.

mdelanno, 2019-03-11:
Yes, it will read the wrong level if we go out of the grid in the X+ direction. In the Y+ direction, it will goes out of the bounds of the array (I've tested it by porting the code to C#). But if you stay in the probed area, even on the edge, it's OK I think. However, there may be problems with rounding.

wolfmanjm, 2019-03-11:
No it is not ok at the edges, it will read the wrong values from the other side of the board.
You can't go out of the bounds as the bounds are clipped to the size of the probe.

wolfmanjm, 2019-03-11:
Can you please test PR #1380

closed 2019-03-12 by wolfmanjm