Smoothieware/Smoothieware #1394: Crash when probing using the panel

unlimitedbacon, 2019-04-18:
#### What were you doing?

1. In the panel menu go to Probe > Z Probe
2. Before the probing finishes, go back to the Watch screen.

#### What did you expect to happen?
Z probing to finish successfully.

#### What happened instead?
Smoothie crashes.

#### Branch & Commit or Version of Smoothieware

```
Build version: edge-de35392, Build date: Apr 18 2019 14:31:09, MCU: LPC1769, System Clock: 120MHz
5 axis
```

#### Hardware model you are using (Make, and version)

    Azteeg X5 Mini V2

#### Host and Operating System running the Host

    None

#### Link to and console logs

* [GDB mini-dump](https://gist.github.com/unlimitedbacon/bf27ab7982781911f3ec76a63519c6b3#file-gdb-txt)
    
#### Link to running config and M503 results

* [config](https://gist.github.com/unlimitedbacon/bf27ab7982781911f3ec76a63519c6b3#file-config)
* [M503](https://gist.github.com/unlimitedbacon/bf27ab7982781911f3ec76a63519c6b3#file-m503-txt)

wolfmanjm, 2019-04-19:
don't do that :) wait until it finishes. FWIW I'll see if there us anything I can do about it, but I doubt it.

wolfmanjm, 2019-04-19:
I looked and there is little I can do to fix this. You hit an edge condition which is easily avoided by waiting for it to finish, but would be very hard to fix.
Basically when you dismiss the screen it is deleted, but the probe has not yet finished when it does it starts executing in a context that has been deleted. We have to delete the context to save memory. This is a trade off between having functionality in a tiny amount of RAM. The only other option is to remove the function, or rewrite the entire probe section.

wolfmanjm, 2019-04-19:
FWIW it is possible to add a flag that indicates if an operation is in progress and if so to not allow the screen to be exited, however I am not sure if you do that how you will actually be able to exit the menu again.

wolfmanjm, 2019-04-19:
try PR #1395 see if that helps, I don't have a panel setup to test on.

unlimitedbacon, 2019-04-19:
That seems to work well enough.

closed 2019-04-30 by wolfmanjm