Smoothieware/Smoothieware #1396: M118 support request

luc-github, 2019-05-03:
Hello,

Marlin and Repetier have now `M118 <MSG>`support, it would be great if Smoothieware support it - I could not find any equivalent in smoothieware GCODE/console command.

```
M118 <message> - Write message to host

echo: <message>
```
I am working on ESP8266/32 project that will use this command to ask ESP8266/32 to send notification using Line or Pushover or email (https://github.com/luc-github/ESP3D/wiki/Notifications-in-3.X). 
Because the ESP board is connected to smoothieboard by serial, if people add something like 
`M118 [ESP600] Print Done`
in their end script, ESP could catch the :
`echo :  [ESP600] Print Done` 
and process the [ESP600] command to send notification. 

This notification feature is already implemented in GRBL_ESP32, and with M118 command soon it will be for Repetier and Marlin, so I think it would be great if Smoothieware has it too.

I hope it can be possible, 

Thank you,

wolfmanjm, 2019-05-07:
M118 is a non compliant gcode and does not follow the NIST rules for gcodes (Letter number only).

So M118 will not be added. However a command like ```echo some message``` is more than welcome as a contribution. (even m118 with  a lowercase m).

closed 2019-05-07 by wolfmanjm

luc-github, 2019-05-07:
Ok I will try to see `echo <msg>` just need to see how to detect all active serial connections

luc-github, 2019-05-30:
Hi @wolfmanjm I did tests using new echo command which work well and I found M1000 command lowercase all commands which may be not expected for `echo` as it change output.   
I worked around by adding here : https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/communication/GcodeDispatch.cpp#L317
```
if (cmd == "echo"){
        args = str;
       shift_parameter(args);
}
```

giving :
```
...
string args= lc(str);
string cmd = shift_parameter(args);
if (cmd == "echo"){
    args = str;
    shift_parameter(args);
}
// find command and execute it
if(!SimpleShell::parse_command(cmd.c_str(), args, new_message.stream)) {
    new_message.stream->printf("Command not found: %s\n", cmd.c_str());
}
...
```
which does not affect current code if not echo command - 
Would you want me to do a PR ? Or would you suggest another approach ?

wolfmanjm, 2019-05-30:
No, just don't use M1000 it has a very special usage and this simply is not needed. M1000 is only used for some hosts that don't allow pass thru commands. Octoprint and pronterface all have ways of doing pass thru.
In general I don't like exceptions like this.

luc-github, 2019-05-30:
Ok great then - thank you