Smoothieware/Smoothieware #1404: Feature request: delay before changing speed multiplier from panel

nmaggioni, 2019-05-19:
I'm using the RDGLCD and from time to time I accidentally bump into the encoder after having exited the menu, thus changing the speed multiplier by a small percentage. It could be useful to add a "delay" between the start of the encoder's rotation and the actual change of the speed multiplier: this way a quick nudge won't affect it.

The inspiration for this feature comes from my CNC running Marlin v2, which ignores slow or brief rotation of the encoder and only begins to change the speed multipler if a fast or sustained rotation is detected.

---

+ **Branch & Commit or Version of Smoothieware:** `Build version: edge-2b729e6, Build date: May 18 2019 04:05:35, MCU: LPC1769, System Clock: 120MHz`, `5 axis`
+ **Hardware model I'm using:** Smoothieboard 5X V1, ReprapDiscount GLCD

wolfmanjm, 2019-05-19:
I am afraid this is unlikely to be added unless someone else wants to submit a PR for this feature. My personal feeling is this would be user error and adding something quite complex to avoid it is overkill. Plus you can simply adjust it back and the effects do not actually take affect immediately anyway as they are delayed by the planner queue, so if you re adjust quickly enough the actual change in feedrate will be negligible. You could also de sensitize the encoder wheel with the various config options already available to do that.

wolfmanjm, 2019-05-19:
FWIW a simpler option to implement would be to add an optional config option to increase the number of clicks required to change just the feedrate. I would accept a PR if someone were to add that.

closed 2019-05-19 by wolfmanjm