Smoothieware/Smoothieware #1407: Feature Request: Porting Smoothie/Smoothieware to the new STM32MP1 MCU series?

Gamester17, 2019-06-26:
Can I suggest that you port Smoothie/Smoothieware to the new STM32MP1 MCU series?
https://blog.st.com/stm32mp1-mpu-stm32mp157a-ev1-stm32mp157c-dk2/

I am only an end-user / technology enthusiast myself and I do not actually know if anyone else is already working on a 3D-printer controller board based on the new STM32MP1 MPU series or not, however reading about it I can not help think that it would fit perfect for 3D-printer controller boards as it combines one or two Cortex-A7 processors @ 650Mhz alongside a very fast Cortex-M4 MCU @ 209MHz. I read that STMicroelectronics claims STM32MP1 architecture enables developers to use the same software as STM32 MCU series, and theoretically, with the STM32MP1 MPU series you could for example run both Smoothie/Smoothieware and a full Linux distribution on the same board. Should be noted though that even the fastest Dual Cortex-A model of STM32MP1 only run at 650MHz per core and supports up to 1GB of external RAM so that part is on paper maybe only twice as powerful as a Raspberry Pi Zero when considering STM32MP1 are based on Cortex-A7 instead of the older ARMv6, but that should still be enough to enable users to run a full Linux distribution with something like OctoPrint or similar smoothly.

The fork Smoohieware CHMT is a previous port to port to STM32F4
https://github.com/mattthebaker/Smoothieware-CHMT

For technical specification on summery on the new STM32MP1 MCU series checkout:
https://www.cnx-software.com/2019/02/21/stmicro-stm32mp1-cortex-a7-m4-mpu/

STM32MP157A-DK1 and STM32MP157C-DK2 Discovery kits are $99 with LCD or $69 without
https://www.st.com/en/evaluation-tools/stm32mp157a-dk1.html
https://www.st.com/en/evaluation-tools/stm32mp157c-dk2.html
STM32MP157A-EV1 Evaluation kit for PCB engineers does have a steeper price at $399
https://www.st.com/en/evaluation-tools/stm32mp157a-ev1.html

First read about this MCU/MPU in the release news about PanGu board from I2Som
https://www.cnx-software.com/2019/06/25/pangu-board-stm32mp1-sbc/

wolfmanjm, 2019-06-26:
Sorry this is not something we would be doing. Smoothie V2 has been ported to a dual core LPC4330.

I know from experience that smoothieware is not portable, and would virtually need a complete rewrite to port to an STM32 CPU.

closed 2019-06-26 by wolfmanjm

arthurwolf, 2019-06-27:
This might be a path towards v3 we'll study in time, thanks for the info.
Cortex-A + Cortex-M is something we've discussed in that context. It's way
too early at the moment though, v2 isn't even hardware-released at the
moment.