Smoothieware/Smoothieware #1414: echo command return anything from an http request

lautr3k, 2019-07-13:
Hi,
I'm not sure if it's normal, then in doubt I open this issue. I work on smoothie-happy v2 and I want to interpret this command correctly.

https://github.com/Smoothieware/Smoothieware/blob/9e5477518b1c85498a68e81be894faea45d6edca/src/modules/utils/simpleshell/SimpleShell.cpp#L463

arthurwolf, 2019-07-13:
So you used the echo command and it does nothing over the network, is what
you are saying ?

lautr3k, 2019-07-13:
@arthurwolf oui, c'est ça ;)
Je pense qu'il manque la ligne `stream->printf("echo: %s\r\n", parameters.c_str());` en doublon de `THEKERNEL->streams->printf("echo: %s\r\n", parameters.c_str()); ` pour que le serveur web retourne quelque-chose.

arthurwolf, 2019-07-13:
Well, stream*s* is supposed to printf to all currently registered streams.
But it's possible that a current http request isn't in that list. I guess
solutions include adding the current request to that list and removing it
when the request ends ( not sure how practical that is ), or do the
stream->printf you suggest IF this is an http request.

wolfmanjm, 2019-07-13:
Correct HTTP requests are not, and should NEVER be, in the streams list.
FWIW echo is a meaningless command and would have no meaning being used in an http request anyway.

closed 2019-07-13 by wolfmanjm

lautr3k, 2019-07-13:
@wolfmanjm Thank you for the precision, sorry I found no information on this function apart by doing archeology in the code ... Sorry to bother you with my questions, but I will continue ;)