Smoothieware/Smoothieware #1415: break function kills the web server before the answer is sent

lautr3k, 2019-07-13:
The break function kills the web server before the answer is sent. This is not a big problem when you know it, but it's annoying for the consistency of the API. 

https://github.com/Smoothieware/Smoothieware/blob/9e5477518b1c85498a68e81be894faea45d6edca/src/modules/utils/simpleshell/SimpleShell.cpp#L688

wolfmanjm, 2019-07-13:
break is for debugging, if you do not understand how it is used then don't use it.

closed 2019-07-13 by wolfmanjm

arthurwolf, 2019-07-13:
I think this is how we want break to work, it's used for debugging so it
should stop the code as soon as possible.

lautr3k, 2019-07-13:
@wolfmanjm Condescension is surely the worst enemy of volunteering... But thanks again for the precision, it's true that I do not know how to use this function, but thanks to you again, now I know it (or not)...