Smoothieware/Smoothieware #1418: Support Prusa's M73 - Show percent done and print time remaining

mvadu, 2019-08-09:
Prusa Slicer has an option to generate [M73](https://github.com/prusa3d/Prusa-Firmware/blob/84cf1b83c3ab476fe72b2c9b416694f7617dc406/Firmware/Marlin_main.cpp#L5827) which adds `M73 Pxx Ryy` at every one minute intervals into g code, where `Pxx` is percentage complete, and `Ryy` is remining time in minutes. This could be really helpful to print in console in pronterface or in ESP3D console monitors. 

Currently I have `echo layer-[layer_num]` in Prusa Slicer layer change gcode, so that I get to see which layer is being printed, but knowing how much time is remaining (as it can be dependent on the complexity of the layers, so 50% layers does not equal to 50% print done) would be really awesome.

arthurwolf, 2019-08-09:
Wouldn't M117 work here ?

mvadu, 2019-08-09:
Two things:
1. Prusa Slicer already calculating the percentages, and remaining times and adding them to gcode with M73. Using it as is will give a straightforward workflow for those of us using Smoothie with Pura Slicer.
2. M117 seems to be done for the printers with display, mine does not have it. I have only access to smoothie console via WiFi. 

Supporting this on console stream will help to integrate with Octa print based workflows (say you want to change the filament at 50%, or getting an alert once the print is done etc). M117 is and will remain to be more "print whatever you want" Gcode, while M73 can be specific to progress reporting Gcode.

arthurwolf, 2019-08-09:
I'm not clear exactly what M73 would do though.

mvadu, 2019-08-09:
How about combination of M117 and echo commands :)  Instead of ` possible_command` in `M117` case, here we will know exactly what to print.

[echo_command](https://github.com/Smoothieware/Smoothieware/blob/9e5477518b1c85498a68e81be894faea45d6edca/src/modules/utils/simpleshell/SimpleShell.cpp#L460)

[M117](https://github.com/Smoothieware/Smoothieware/blob/2b84ffc6d15f8f98533a39a943cde84aebd87504/src/modules/communication/GcodeDispatch.cpp#L294)

mvadu, 2019-08-09:
Oh, it would have to print to streams instead of just to panel.

wolfmanjm, 2019-08-12:
A work around (and probably something you could do easier) is to write a post processor for your gcode that extracts all M73 lines and replaces them with echo lines translating the parameters to english.

mvadu, 2019-08-12:
I did a poor man's version with a simple replace all macro in Notepad++, but keep forgetting to run it before uploading it to printer. Hence took time to open this issue here 😁

arthurwolf, 2019-08-12:
Slic3r lets you run scripts when a Gcode file is generated ( essentially a
post-processor like used on CNC mills ), doesn't Prusa's version keep that
feature?

mvadu, 2019-08-12:
I have not explored that option yet Arthur. Let me check that tonight if it does then I don't think I need this change in Smoothie. `echo` does the job

ArchConMan, 2019-08-12:
Is that not the suggested "correct" way to modify any G-code?
Redefining codes for custom use is wrong/rude but "post-process" as you
please.
Maintain standards while accepting scripting power is The G-code.

compares Firmware Gang territories by color
https://reprap.org/wiki/G-code

mvadu, 2019-08-18:
@ArchConMan as per your linked page [`M73`](https://reprap.org/wiki/G-code#M73:_Set_build_percentage) is already supported by Marlin, MakerBot and MK4duo flavors. Unfortunately there is no ISO or W3C kind of organiztions working to define what is "golden standard" for g codes. All I was asking for was Smoothie to support what I thought was a good use case. That being said, thanks to @arthurwolf for the post processing script suggestion. Original Sli3r developed in Perl expects an executable as post processing script, buit Prusa version supports running any command including powershell. So to solve my original ask, I have this one liner now, which converts `M73 P0 R26` to `echo Completed:0%, 26 minutes left`. 

Here is my one liner which I have tested in Prusa Slicer 2.0
```Powershell
"c:\windows\System32\WindowsPowerShell\v1.0\powershell.exe" -Command "& { (resolve-path \"$($args -join ' ')\" | get-content | ForEach-Object { if($_ -match \"^M73 P(?"<percent>"\d+) R(?"<timeleft>"\d+)$\") {Write-Output \"echo Completed:$($Matches.percent)%, $($Matches.timeleft) minutes left\"} else {write-output $_}}) | out-file $($args -join ' ')}"
```

,

closed 2019-08-18 by mvadu