Smoothieware/Smoothieware #1421: A Bug in the "MorganSCARASolution.cpp"  to reopen

svsergo, 2019-08-20:
Cartesian_to_actuator function gives incorrect values, if arm1_length is not equal with arm2_length

void MorganSCARASolution::cartesian_to_actuator(const float cartesian_mm[], ActuatorCoordinates &actuator_mm ) const
{

....
if (this->arm1_length == this->arm2_length)
SCARA_C2 = (SQ(SCARA_pos[X_AXIS]) + SQ(SCARA_pos[Y_AXIS]) - 2.0f * SQ(this->arm1_length)) / (2.0f * SQ(this->arm1_length));
else
# incorrect

    SCARA_C2 = (SQ(SCARA_pos[X_AXIS]) + SQ(SCARA_pos[Y_AXIS]) - SQ(this->arm1_length) - SQ(this->arm2_length)) / (2.0f * this->SQ(arm1_length));
# should be instead
SCARA_C2 = (SQ(SCARA_pos[X_AXIS]) + SQ(SCARA_pos[Y_AXIS]) - SQ(this->arm1_length) - SQ(this->arm2_length)) / (2.0f * this->arm1_length * this->arm2_length);
....

wolfmanjm, 2019-08-20:
@qharley any comments?

qharley, 2019-08-20:
I will take a look.

qharley, 2019-08-20:
Yes, I sent a pull request for this if it has not been done yet.
My machines always have the arms the same length... that's how we missed
that.

closed 2019-08-20 by wolfmanjm