Smoothieware/Smoothieware #1424: KillButton Bug

fishpepper, 2019-09-04:
The following situation: A smoothieboard running in corexy mode with corexy homing enabled (maybe not relevant to the bug).
One of the standard mushroom shaped Estop buttons is connected to a pin and configured as kill button.
And yes, I know the software "kill" feature is not a real Estop but my machine is harmless with very weak motors... There is no need to kill the motor driver power as on my real cnc machine ;)

Situation 1:
On Smoothie startup this button is pressed down (and stays down).
When sending a homing command the smoothieboard leaves the kill state.
From now on everything works as the button is not pressed.

Situation 2: 
Pressing ESTOP while running (stays down).
Kill is triggered, sending another home command will leave the kill state and the pressed button is ignored until unpressed and pressed again.

Reason:
It seems like the homing unsets the kill command.
Either the src/modules/utils/killbutton/KillButton.cpp statemachine has a bug or 
Endstops::process_home_command(Gcode* gcode) should check if the endstop is pressed before unsetting THEKERNEL->is_halted();

Statemachine states:
Button Pressed:
IDLE -> KILL_BUTTON_DOWN
once killed bit is set
KILL_BUTTON_DOWN -> KILLED_BUTTON_DOWN
stays here because button is pressed down:
KILLED_BUTTON_DOWN 

If the homing now unsets the kill state while the button is pressed,
the statemachine will stay here without retriggering.

I guess the optimal fix would be that the homing does not leave the kill state when the button is pressed. I could not figure out where exactly this happens during homing.

I think the best place to check for the kill button still pressed is the
Kernel::call_event(_EVENT_ENUM id_event, void * argument) function.
In any situation, this should check the kill button when releasing halted? 
I was thinking I ask you before I submit a patch for that.

wolfmanjm, 2019-09-04:
Kill is meant to be triggered by a normal normally open button not an estop. Use your estop to cut power not send a kill.

However there is a kill button setting that may work for you...

`kill_button_toggle_enable true`

Try it, if it works great, if not then replace your estop with a normal non latching button.

closed 2019-09-04 by wolfmanjm

fishpepper, 2019-09-04:
No, kill_button_toggle_enable does not work either.

Eeven if a normal push button is used, the halted condition should not be removed when the button is still pressed?

wolfmanjm, 2019-09-04:
If you don't want home to clear the HALT state then don't use $H, use G28 (or if in grbl mode use G28.2). Only $H clears HALT to be GRBL compatible.

fishpepper, 2019-09-04:
Thanks, now I see where the halt is cleared. In case anyone else wonders:
https://github.com/Smoothieware/Smoothieware/blob/4fd36326b454e2923ee6fa2511265cbbe76053de/src/modules/utils/simpleshell/SimpleShell.cpp#L242

But from the users point of view I would not expect that behaviour.
Clearing the halt state should not be possible while the kill button is pressed. 
Could be a latching EStop or even a broken (shorted, blocked) normal push button.

I will submit a patch when I am at home in the evening.

wolfmanjm, 2019-09-04:
Your patch will not be accepted so I would not bother to post it, but thanks. Your proposed changes are not acceptable.

fishpepper, 2019-09-04:
I don't get your point. Why is it acceptable to clear a flag while the cause of the flag is still active?
This sounds wrong to me. It feels like a circuitbreaker that can be switched on (and then stays on) while the short circuit is still in place.
I will put a fix in my fork. Maybe its useful to others.

wolfmanjm, 2019-09-04:
We disagree on that. 
Initially kill buttons were ONLY supposed to be momentary buttons, so extending to an estop is new functionality (and a misuse of the estop IMO).
Clearing the Halt state takes precedence over the button in that case as well, if it is still enabled then don't issue the clear halt command.
There are several ways to clear HALT, $H is only one of them, and your proposed change probably won't cover all the other methods, and I do not see a way to do that cleanly. Your proposal breaks the modularisation, as kill buttons can ONLY be handled in the kill button module, and cannot be accessed anywhere else in the code base.
The only thing I may consider is if the Kill button FSA is modified to reassert HALT if toggle is enabled and the button is still down and HALT has been triggered and since cleared. Anything else would be incorrect.

fishpepper, 2019-09-04:
You are right, extending this to an estop button is a new functionallity.
But the underlying cause is not. A broken, always pressed kill button will never trigger the halted command. My patch is in #1425. Not sure if this is ugly or not, i am only using functionallity that exists and is used elsewhere (getter for public data).

Another option might be subscribing to ON_HALT and setting it again (ugly in my opinion).

If both are to ugly for you I could have a look into changing the FSM.

wolfmanjm, 2019-09-04:
TBH I ran into this exact issue years ago when I wanted my estop to both cut power and issue kill, and I never found a clean way to deal with the issue. So I have looked into trying to solve this in the past and decided the code required to solve it was not worth the effort as simply not using a latching kill switch solved the issue, or not issuing unkill while latched.
Both proposed solutions, although they would work are too ugly to contemplate. The correct solution is to modify the FSM to handle the case when toggle is set. However having looked into that solution as well I found it complicated the FSM to the point it could not be easily tested for all conditions, and made the code almost unreadable, so again not worth the effort.

I suggest you keep this in your own fork if you insist on using a latching estop. As I said this is not a bug as the kill switch is expected to be a momentary switch so should not be closed when unkill is issued.

wolfmanjm, 2019-09-04:
Additionally I see a use case (especially in my use case where the estop cut power and triggered HALT), where I would want to force a clear of the HALT (usually with $X) but leave power cut, and this fix would not allow me to do that unconditionally. $X should always IMO clear the HALT if it is explicitly issued.

fishpepper, 2019-09-04:
Its to late for testing it on the machine, but this could be a much easier fix to the issue.
But it is untested, maybe i got the control flow wrong and this does not work. I will test it tomorrow.
```
@@ -89,7 +89,13 @@ uint32_t KillButton::button_tick(uint32_t dummy)
                 if(killed) state= KILLED_BUTTON_DOWN;
                 break;
             case KILLED_BUTTON_DOWN:
-                if(this->kill_button.get()) state= KILLED_BUTTON_UP;
+                if (this->kill_button.get()) {
+                    state= KILLED_BUTTON_UP;
+                } else if ((toggle_enable) && (!killed)) {
+                    // button is still pressed but the halted state was left
+                    // re-trigger the halted state
+                    state= KILL_BUTTON_DOWN;
+                }                    
                 break;
             case KILLED_BUTTON_UP:
                 if(!killed) state= IDLE;

```

wolfmanjm, 2019-09-04:
Yes that would be much cleaner if it works. I'm not sure the logic is totally correct, but it is in the right direction.

fishpepper, 2019-09-05:
I just tested it, it works. 
When homing it starts for a brief period of time (because the statemachine is running at 5(?) Hz or so).
Then the halted state is triggered again and the homing stops. Not 100% clean but at least it does no longer get stuck in the FSM.

MAke sure to configure the kill button as toggle in the config file to make this work.

arthurwolf, 2019-09-05:
Thank you so much for insisting and working through to making it work with
the rest of Smoothie etc, this is really cool.