Smoothieware/Smoothieware #1427: difference between !/~ and suspend/resume

fishpepper, 2019-09-05:
Not so much an issue, more an understanding problem.

I am currently adding the host actions to cncjs so that a suspend button press will stop cncjs from sending more data.
The part with reading the actoin:suspend smoothie message works fine. I can stop  the print.

However clicking on resume in cncjs sends a ~. This starts the machine again, but does not clear the suspended flag. 
Resending a suspend command says the machine is still suspended.

What is the expected behaviour? Should ~ also clear the suspend state?

arthurwolf, 2019-09-05:
Hey, ideally questions like this you want to ask on IRC, the mailing lists,
the forums, social media, etc, we try to keep issues focused on a very
specific task.

wolfmanjm, 2019-09-05:
do not use either ! or ~ as they are not implemented properly and are not supported. They are supposed to be feed hold and resume but don;t actually do anything, instead the host should simply stop sending gcode.

wolfmanjm, 2019-09-05:
Also there is absolutely NO relation between M600 (suspend) and M601 (resume) and !/~
They are two totally different operations. (if ~/! actually worked as they are really grbl things that we do not support).

closed 2019-09-05 by wolfmanjm

fishpepper, 2019-09-06:
Ok good to know that this is not supported. Then I will think about a patch for cncjs. This uses ~/! for pause and resume.

Maybe the Smoothieware documentation should be updated, this site:
http://smoothieware.org/stopping-smoothie
lists !/~ as supported commands.
From that page it was not clear to me if why there is a difference. This is why I opened this issue ;)