Smoothieware/Smoothieware #1432: Firmware crashes on malformed config-get request.

fossum, 2019-09-28:
#### What were you doing?

I hadn't started anything. Connected over telnet and the web page up in the background. I recently updated to the latest edge firmware, but I think this existed before that.

> \> help
> Commands:
> ...
> config-get [<configuration_source>] <configuration_setting>
> ...
> \> config-get
> [[[frozen]]]

#### What did you expect to happen?

I was trying to get a more detailed usage example of the command, but I would have also expected an error.

#### What happened instead?

The firmware froze and had to be restarted.

#### Branch & Commit or Version of Smoothieware

> \> version
> Build version: edge-2b84ffc, Build date: Jul  3 2019 21:31:48, MCU: LPC1769, System Clock: 120MHz
> 5 axis

#### Hardware model you are using (Make, and version)

Smoothieboard 5X V1a

#### Host and Operating System running the Host

Telnet+Smoothie Web, linux x64.

#### Link to and console logs

Pasted in other sections.
    
#### Link to running config and M503 results

##### M503

> ; config override present: /sd/config-override
> ;Steps per unit:
> M92 X80.00000 Y80.00000 Z1600.00000 
> ;Acceleration mm/sec^2:
> M204 S3000.00000 
> ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
> M205 X0.05000 Z-1.00000 S0.00000
> ;Max cartesian feedrates in mm/sec:
> M203 X500.00000 Y500.00000 Z5.00000 S-1.00000
> ;Max actuator feedrates in mm/sec:
> M203.1 X1250.00000 Y1250.00000 Z62.00000 
> ;Digipot Motor currents:
> M907 X1.50000 Y1.50000 Z1.50000 A1.50000 
> ;E Steps per mm:
> M92 E140.0000 P57988
> ;E Filament diameter:
> M200 D0.0000 P57988
> ;E retract length, feedrate:
> M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
> ;E retract recover length, feedrate:
> M208 S0.0000 F480.0000 P57988
> ;E acceleration mm/sec²:
> M204 E500.0000 P57988
> ;E max feed rate mm/sec:
> M203 E50.0000 P57988
> ;PID settings, i_max, max_pwm:
> M301 S0 P66.8520 I8.7104 D128.2723 X255.0000 Y255
> ;Max temperature setting:
> M143 S0 P300.0000
> ;PID settings, i_max, max_pwm:
> M301 S1 P117.7520 I10.7782 D321.6103 X255.0000 Y255
> ;Max temperature setting:
> M143 S1 P300.0000
> ;Home offset (mm):
> M206 X0.00 Y0.00 Z0.00 
> 
> ok

##### Configuration

https://pastebin.com/B382hR2V

wolfmanjm, 2019-09-28:
config-get is not really supported, when used like this it tries to load the config cache which quite often will run out of memory and crash. I leave it in there as it is sometimes useful for development and if used in a limited config system in development sometimes has enough memory to actually work.

wolfmanjm, 2019-09-28:
BTW this fact is documented in the wiki.

closed 2019-09-28 by wolfmanjm

fossum, 2019-09-29:
Yeah, I know it can crash, but the command was malformed. Seems it should have returned without loading the configuration.

I did know of the documentation.

arthurwolf, 2019-09-29:
If you need to edit the config, and you are talking over the network, the
way I do it in the web interface is : http download the config file // edit
value // http upload // http command reset
Works like a charm. If you need a link to the code tell me.

fossum, 2019-09-29:
@arthurwolf that sounds perfect. I'd love to get that link.

I'm mostly trying to edit settings on my computer in a Git repo and upload them via Telnet, which I hadn't gotten working. Not sure how any Telnet transfer is working actually (I think that freezes my board to). If I can do everything over HTTP, I think I'd like to try that.

arthurwolf, 2019-10-02:
I think this is the simplest version of it I have online right now:
https://github.com/Smoothieware/Webif-pack/blob/gh-pages/new-web-interface/functions.js#L158