Smoothieware/Smoothieware #1434: Switch commands pauses

lamest, 2019-10-28:
#### What were you doing?

I'm trying to use a switch like that:

```
M280 S40
M282 S40
G1 X37.001 C0.000
X147.000
X207.000 C315.000
Y-14.001 C270.000
Y-119.999
M280 S38
M282 S42
G1 Y-239.999 C180.000
```

#### What did you expect to happen?

expecting that between `Y-119.999` and `G1 Y-239.999 C180.000` the machine would not stop or pause.

#### What happened instead?

pauses every time when there are M280 or M282 with changed parameters.
note: it's not pauses when I replace M280 S.. and M282 S.. with M114.
it's not pauses when there are M280 S.. and M282 S.. with the same S value.

no-pause examples:
```
M280 S40
M282 S40
G1 X37.001 C0.000
X147.000
X207.000 C315.000
Y-14.001 C270.000
Y-119.999
M280 S40
M282 S40
G1 Y-239.999 C180.000
```

```
M114
M114
G1 X37.001 C0.000
X147.000
X207.000 C315.000
Y-14.001 C270.000
Y-119.999
M114
M114
G1 Y-239.999 C180.000
```

#### Branch & Commit or Version of Smoothieware

commit id = 4fd36326b454e2923ee6fa2511265cbbe76053de

#### Hardware model you are using (Make, and version)

   MKS SBase v1.3

#### Host and Operating System running the Host

    OctoPi Linux, OctoPrint

#### Link to and console logs

* will add later
    
#### Link to running config and M503 results

https://gist.github.com/DasRegal/c33ca7a6df2e86febaff274fef057451

`; No config override;Steps per unit:M92 X64.00000 Y64.00000 Z2560.00000 A100.00000 ;Acceleration mm/sec^2:M204 S400.00000 Z60.00000 A600.00000 ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:M205 X0.05000 Z-1.00000 S0.00000;Max cartesian feedrates in mm/sec:M203 X200.00000 Y200.00000 Z150.00000 S-1.00000;Max actuator feedrates in mm/sec:M203.1 X200.00000 Y200.00000 Z39.00000 A50.00000 ;WCS settingsG54;Digipot Motor currents:M907 X0.80000 Y0.80000 Z1.50000 A1.50000`

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2019-10-28:
This is correct behavior the pause is required to make sure the switch is synchronized with the surrounding gcode. If it did not pause the switch would happen about 32 gcodes before, as it would be executed immediately rather than in the correct sequence.

closed 2019-10-28 by wolfmanjm

arthurwolf, 2019-10-28:
To go into more detail, if we were to implement a solution for switches to
be operated without any pauses, it'd have immense costs in terms of
processing power/timing ( ie smoothie would need to be much much slower in
it's execution ). Smoothie is optimized for movement, switches are more of
a secondary feature, it's the first time I years somebody
notices/supposedly needs switches to be more integrated with the flow.

lamest, 2019-10-30:
@arthurwolf thanks for the explanation. We need this because we use a lot of switches and use them frequently. Also, we have slow accelerations. All of this is leading to bad printing quality.

triffid, 2019-10-30:
It's possible to queue switch actions (instead of flushing the queue), but
smoothie doesn't do it because it actually breaks the majority of usage
cases.

You could create a local branch where they're queued if you like

wolfmanjm, 2019-10-30:
@triffid It is no longer possible to queue non move commands, as they were executed in ISR context which is a no no, all hooks for that were stripped out a while ago.
This would be a very complex thing to add in order to avoid commands being executed in the step generating ISR. It is certainly possible but not trivial.

arthurwolf, 2019-10-30:
In a custom branch, one could make a special version of the switch code
that switches bits on each block, and then have custom code that
reads/applies this to pins in the step ISR. I think that'd work, if you
need help implementing something like that don't hesitate to ask me.

lamest, 2019-11-27:
@arthurwolf we need your help for sure. How can we do that?