Smoothieware/Smoothieware #1446: Feature: support for clustering to boost engraving/move speeds

dlarue, 2020-03-11:
Cohesion3D has added clustering support to their fork(?)  of Smoothieware which they tested to triple engraving speeds to match or best GRBL-LPC speeds. Is this on the Smoothieware release timeline or not even something being considered?

arthurwolf, 2020-03-11:
Can you describe in more detail what you mean by clustering?

wolfmanjm, 2020-03-11:
currently not even being considered. sorry. however we would welcome a contribution if you want to add it.

dlarue, 2020-03-11:
I've not found any source code on this, only a mention first on the LightBurn forum and then on the Cohesion3D forum where a firmware image was provided. It looks like Ray worked with Oz on this and obtained some faster rastering/movements. I'll try to dig more into it.

https://forum.cohesion3d.com/t/improved-raster-speed-firmware-update-for-cohesion3d-boards/850

wolfmanjm, 2020-03-11:
yea this is typical of them, not contributing to open source and keeping their changes proprietary violating the license.

wolfmanjm, 2020-03-11:
FWIW this probably only improves windows streaming performance, as it has a broken USB serial timing, whereas Linux seems to stream just fine and fairly fast. I don't use windows so can't really help. TBH I am skeptical and always have been about the reasons for compressing the gcode.

dlarue, 2020-03-11:
Here is how Oz explains what it does and it appears to be related to how far ahead Smoothie can look and react in raster mode so I would think it'll help Linux users too(myself included):

https://forum.cohesion3d.com/t/improved-raster-speed-firmware-update-for-cohesion3d-boards/850/8

wolfmanjm, 2020-03-11:
well I can stream raster to my K40 at 100mm/sec real speed. Which seems fast to me. (K40 certainly is not stable at that speed).

dlarue, 2020-03-11:
They tested at 250mm/sec with 254 dpi and ran tests on other firmware showing the improvements are significant. I agree, you'd better have your acceleration and deceleration tuned or your laser will walk off the table and onto the floor.

wolfmanjm, 2020-03-11:
Yea I was aware of this hack, He basically sends an illegal gcode string to smoothie which compresses a scan line into a single gcode. It is quite clever, but certainly breaks all the NIST rules for GCode.

dlarue, 2020-03-11:
Ah, so cool trick/hack but non-standards based.  Thanks for looking into it and replying.

closed 2020-03-11 by dlarue

wolfmanjm, 2020-03-11:
I think I did talk to the lightburn guy about it some time ago, and I was not really comfortable with the hack TBH. So I guess they did it anyway and stuck it in a non source code proprietary bin for c3d. I suppose we could point out that violates the GPL, but I don't think they care.

arthurwolf, 2020-03-11:
He shares the code in a dropbox link lower in the forum thread. I asked him
to do that in a normal/visible way.

LightBurnSoftware, 2020-05-17:
If you aren't going to take the code, it seems a bit pointless to submit it.  It's not NIST standard, but it's the only way to get Smoothieware to process the instructions fast enough to break the 1000 gcode/sec limit, and it is backward compatible with normal gcode.  It takes less moves in the plan buffer, and allows the controller to see up to 8x farther ahead than normal when processing raster data with basically no negative side effects, so I don't understand the reluctance.

arthurwolf, 2020-05-17:
It's not pointless. It'll be in the list of third-party branches on the
wiki, people would test it and give us feedback, and it'd help making
progress towards whatever we'll end up doing about this ( if we do anything
).
Sharing the code is definitely a better idea than not doing so... If you
don't share the code, nothing can happen ...

LightBurnSoftware, 2020-05-18:
... but I did share it - I emailed it directly to Jim, who looked at it and decided it wouldn't be included.  The code is posted on the C3D and downloadable by anyone.  If you choose not to take it, your loss. (well, your users loss, but I digress)

dlarue, 2020-05-18:
As I mentioned earlier, Ray did some tests and the speed improvements were significant.  Could have been only on Windows but while I loathe that platform, lots of people use it so there are benefits.

wolfmanjm, 2020-05-18:
Unfortunately when you send me the code I am unable to do a proper review as I have no diffs. By submitting a PR I can see exactly what lines have changed and what files have changed. It also allows for others to review the changes as well, not just me (as I may be a little biased at times).
As for not being NIST compliant we do have a way of getting around that... Most non compliant commands are issued as a lowercase command, which can have any format and any parameters (See the PR for the TMC drivers). It may still be rejected in the exact form it is but it is really hard to say without seeing diffs. 
My main concerns is large changes to core parts of Smoothie can destabilize the entire code base, and so needs careful scrutiny.
It is up to you if you want to submit a PR or not.

arthurwolf, 2020-05-18:
> ... but I did share it - I emailed it directly to Jim, who looked at it
> and decided it wouldn't be included. The code is posted on the C3D and
> downloadable by anyone. If you choose not to take it, your loss. (well,
> your users loss, but I digress)

You need to submit a PR like all other third-party code/branches.

hypoxic, 2020-12-17:
If anyone else is curious, as I was, I merged in the Cohesion3d changes and did a binary diff until the resulting firmware matched. 
https://github.com/hypoxic/Smoothieware/commit/1b14b3e0d18fa9b17c9031bbc5cf6a4c73206764

dlarue, 2020-12-17:
I only see 2 files changed and they are text files( config and readme ) so surely there were code changes too.
But from Jul 2019 this was the list of Smoothieware files changed to implement clustering in the Cohesion3D fork.
https://www.dropbox.com/s/bk44snjpu03dju1/SmooChanges.zip?dl=0

hypoxic, 2020-12-17:
My apologies, the changes are merged on the prior commit if you want to see the diffs. Everything matched their dropbox except for there was a static value that was changed. Compile instructions are in the readme. 
https://github.com/hypoxic/Smoothieware/commit/8b5e9e787185547f7e6145abfaff364093135b5a