Smoothieware/Smoothieware #1450: SCARA home_z_first error

alfie0, 2020-03-31:
Latest firmware version

arm_solution     morgan SCARA
scara_homing     true
home_z_first       true

X and X axis always start to move first.

Running G28 Z0, X-Y axis will also move together first.

wolfmanjm, 2020-03-31:
Scara has a different homing protocol, so I suspect home z first does not apply  to it.
contact @qharley for more details, he handles the scara stuff.

closed 2020-03-31 by wolfmanjm

qharley, 2020-03-31:
Homing in SCARA defaults to all moving together.  I will check if there are
differences between my branch and Smoothie/edge

qharley, 2020-03-31:
Try my branch for SCARA bots.  There are some features that wil not be
incorporated into Smoothie/edge that may come in handy.
I changed the homing routine to bring back the "Home All" function.

qharley, 2020-03-31:
https://github.com/RepRapMorgan/Smoothieware

qharley, 2020-03-31:
We are in lockdown here in South Africa.  I will try to work on an update to the SCARA code that will hopefully please the Smoothie mods

alfie0, 2020-04-01:
I try G28 Z0 in detail, XYZ axis does moving together, but the X axis moves very slowly. I hope that the Z axis can be homing alone. Thank you.

qharley, 2020-04-01:
I just checked.  If you do G28 Z in MorganSmoothie branch it will only home the Z-axis

alfie0, 2020-04-02:
Try different settings and firmware are not good, G28 Z axis and X-Y axis always move together, do not know what reason.