Smoothieware/Smoothieware #1452: No generic way to execute commands from modules other than SimpleShell using G or M-code.

0bo0, 2020-04-06:
#### What were you doing?
    
    M1000 won't execute commands that are not present in SimpleShell::commands_table, 
    like "fire" from laser module. Laser module won't execute it too, because it finds "M1000" string
    at the beginning of the line. So it is not possible to execute that command on hosts that support 
    only G/M codes. It would be useful as M3 doesn't turn the laser on if there's no movement.

#### What did you expect to happen?
 
    I'd expect there's a way for hosts which only support G/M codes to
    execute commands from modules other than SimpleShell. Example: M1000 fire 10

#### What happened instead?

    Error returned: "Command not found: fire", see below.

#### Branch & Commit or Version of Smoothieware

    latest edge

#### Link to and console logs

      Commands typed manually over serial:
    fire 10
    WARNING: Firing laser at 10.00% power, entering manual mode use fire off to return to auto mode
    fire off
    turning laser off and returning to auto mode
    M1000 fire 10
    Command not found: fire
    ok
    M1000 fire status
    Command not found: fire
    ok

      Laserweb:
    M1000 laser 10
    Command not found: laser
    fire 10  
      (comment: laser turns on but nothing works after that, commands not being sent/received
      anymore, likely problem with laserweb not parsing response correctly, 
      maybe expects 'ok' at the end. Would work well with M1000 as it adds 'ok' to the response)

      RepetierHost:
    22:47:58.561 : N20 M1000 Fire Status*44
    22:47:58.562 : Command not found: fire
    
#### Link to running config and M503 results

    https://pastebin.com/p1E82s23
    https://pastebin.com/u37M59gG

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2020-04-06:
I'd suggest using a different host that can send commands. Like Octoprint, bCNC, pronterface, Smoopi.

This is not a bug in Smoothie IMO.

closed 2020-04-06 by wolfmanjm

arthurwolf, 2020-04-06:
Hey there.

We tend not to build hacks into Smoothie *only* because a Host can't do
it's job properly. I recommend you open an issue with your host program, it
should be capable like the others are.

This is.
1. Trivial for the host programmers to fix
2. Hard for us to fix, and would degrade the quality of the code/project

I think this is a bad idea.
Thanks for trying to improve things though.

Cheers.

0bo0, 2020-04-06:
I'm not saying it's a bug. It's a missing feature. Switching to hosts optimized for completely different tasks (I'm using LaserWeb and LaserGRBL currently), just because of this one thing doesn't make any sense.
Even though it's easy to implement this feature on hosts, it wouldn't be needed if smoothieware supported such basic functionality as switching laser on with M3 or M4 while stationary in the first place. And it would be compatible with all existing hosts. Smoothieware is not the most popular firmware on the market so expecting everyone else to adjust is rather naive.

I've just added a special M-code for this on my own as I was missing this feature a lot.

arthurwolf, 2020-04-07:
> I'm not saying it's a bug. It's a missing feature.
It's not, it's hosts not properly implementing compatibility.

> Switching to hosts optimized for completely different tasks (I'm using
> LaserWeb and LaserGRBL currently), just because of this one thing doesn't
> make any sense.
Which you should tell the host devs.

> Even though it's easy to implement this feature on hosts, it wouldn't be
> needed if smoothieware supported such basic functionality as switching
> laser on with M3 or M4 while stationary in the first place.
That is not functionality, that is out-of-standard behavior. The reason
some systems use M3/M4 for laser is simply because that's what's used for
spindle control, and they pretty much don't have any laser-specific
functionality, forcing users to use spindle control for laser control.
This is a hack, not proper functionality.
And it's a very bad idea designing things this way. What if a machine has
both a laser and a spindle? Now you can't have both if M3 is the way you
control your laser. Terrible idea. Shouldn't be the standard ( and it isn't
the standard. for good reasons ).

> And it would be compatible with all existing hosts. Smoothieware is not
> the most popular firmware on the market so expecting everyone else to
> adjust is rather naive.

Smoothie has been ( by far ) the first system to implement most features
related to this. Therefore it's set the standard of how to do things. Hosts
not respecting standards isn't our fault or responsibility.
And we shouldn't be expected to lower the quality of our code, to
compensate for host developers not doing their job ( especially when that
work is extremely trivial ).

> I've just added a special M-code for this on my own as I was missing this
> feature a lot.