Smoothieware/Smoothieware #1453: Command buffer

entolium, 2020-04-09:
I'm writing a PC program to control my cnc with smoothieboard, and I have a big problem streaming gcode blocks.
When the buffer is full, I must wait until the "ok" response.
This is a problem because the program is continously locked waiting.

In GRBL, I can get the buffer space for new blocks (with the "?" command) and send the exact number of blocks to fill the buffer. My program is free to update other information because the serial port is not locked waiting a response.

If this is not possible in smoothie, I think a solution could be respond "no ok" or "full" when the buffer is full and ignore the last block received.

Thank you and sorry for my english.

arthurwolf, 2020-04-09:
Waiting for OK is the correct way to do this. All other ways are incorrect
and *will* break things. There is *absolutely definitely* a way to do this
non-blocking no matter what you are doing your program with, you should
search and find what that way is. Can you tell us more about your host
program ?

wolfmanjm, 2020-04-09:
this is really not an issue, just follow th eping pong protocol as described in the wiki and by various other hosts.

It is essential you send ONE gcode and wait for the ok before you send another. There is no exception to this rule.

You can look at the smoopi source code which is a host I wrote specifically designed to talk to smoothie.

closed 2020-04-09 by wolfmanjm

wolfmanjm, 2020-04-09:
To be more specific, smoothie is nothing like GRBL not even close. There is no Serial buffer as the port is a USB CDC ACM port. It has its own flow control and checksum resend so you don't need to do that, bit there is no serial buffer, and due to the USB flow control if you send more than you should and not wait for ok you can dead lock in your host as you discover.
This will never happen if you follow the ping pong protocol as mentioned.

entolium, 2020-04-09:
I send one block. When receive "ok", send another. When receive "ok" send another.
If not response...what happend?... smoothie is busy? buffer is full? the communication is lost?
If send other command and receive "ok", this "ok" is for the command or for the block pendant?

In a communication, always must respond to a query (ok, no ok, error....). This is basic.
GRBL has the same fault, but I can solve because I can read the buffer status. In smoothieware is not possible.

arthurwolf, 2020-04-09:
This isn't a fault, this is how the protocol is designed to work. It's used
by many software/hosts, it's used by many controllers, it's a standard and
it works very well.

arthurwolf, 2020-04-09:
If you don't get an answer, it simply means you must wait before sending
another. That's the protocol.

entolium, 2020-04-09:
I must wait, yes, I'm doing this, but how long?.
Please, think about this. Is a problem easy to solve.

arthurwolf, 2020-04-09:
> I must wait, yes, I'm doing this, but how long?.
Until you get an answer. Which you will, for sure. It's very simple.

> Please, think about this. Is a problem easy to solve.

This is not a problem, it's a protocol that has worked for a decade across
dozens of firmwares and host programs.
Apparently it's not convenient in your specific case, or it "feels"
different from what you are used to, but there are *millions* of users of
this protocol, and we never get any complaint, including talking every year
to dozens of people developing software talking to this protocol.

entolium, 2020-04-09:
This protocol works because nobody change it, and hosts adapt to it with difficulties.

I think could be much better, for example in this issue. If you don't..."ok"

Thank you very much for your fast answers.
Bye.

wolfmanjm, 2020-04-09:
You need to think about the issues here. The time an ok can take is indeterminate, think... if you send a G0 X100 F10 and it was the last move of say 32 moves which filled up the look ahead planner, then the ok would take at least 10 minutes! so there is NO WAY to say how long it will take and no way for you to know ahead of time and NO WAY for smoothie to tell you. The reason this protocol has been around so long is that it works! If for some reason you have to know how long each one will take then you will need to parse all the gcode yourself and model the planner (with acceleration and deceleration) and basically simulate a motion control system to determine how long it will take.

entolium, 2020-04-15:
https://github.com/LaserWeb/LaserWeb4/wiki/2.2-Initial-Configuration---GRBL-LPC-1.1

_We have ported the ingenious GRBL 1.1 to LPC1769 boards (like Smoothieboard, Cohesion3D, Azteeg X5, MKS SBASE) because Smoothieware has **no decent realtime streaming protocol**, which is needed for fast raster engraving with **realtime control**._

arthurwolf, 2020-04-15:
>
> *We have ported the ingenious GRBL 1.1 to LPC1769 boards (like
> Smoothieboard, Cohesion3D, Azteeg X5, MKS SBASE) because Smoothieware has
> no decent realtime streaming protocol, which is needed for fast raster
> engraving with realtime control.*
>

They are absolutely wrong that this is required to do fast engraving. There
are other ways to achieve it, some very simple, they are just not their
preferred way, because they are used to alternatives ( similar to your
situation ).

There has been disagreement with the Laserweb team about the best
approaches to laser optimization for years ( going way beyond just
communication protocols ), this is just one of many many jabs our way
they've done over the years. They are completely unfair in their
evaluations ( and ignore our remarks and answers even when they have been
told to them many times over the years ), and we tend to have a policy of
ignoring them not to stir discord ( which, like ... thanks for making that
situation worse. I guess you didn't know about the history there, so that's
mostly fine ).
Smoothie's protocol is fine, it works great, it can accomplish anything you
want to achieve, and *does* in other people's projects, you just don't like
it because it's not what you "expected" it to be, and maybe because
smoothie's protocol isn't what would have caused the least amount of effort
on your part.