Smoothieware/Smoothieware #1456: 4th axis stops working when kill button is configured

PKav89, 2020-04-18:
#### What were you doing?

I compiled Smoothieware for 4-axis setup as described in http://www.smoothieware.org/6axis
Flashed it on BigTreeTech SKR v1.3 Board.
Created a 4-axis configuration file.
Worked with machine for several days with no issues.
Added a big red kill button (normal closed), hooked to P2.0 pin.
Added a kill button configuration to a config file.

#### What did you expect to happen?
Machine homes and works as previously.
#### What happened instead?
Machine homes only X, Y and Z axis. A axis does not move, driver stays disabled, so firmware goes into HALT state soon after G28. But A axis endstop works, so if I press it manually, firmware homes, does a bump and think it is in homing position for all 4 axes.
Kill button works as described in http://smoothieware.org/killbutton
If I comment out kill button configuration lines, machine homes all axes correctly and works as expected, but, of course, kill button does not work.
#### Branch & Commit or Version of Smoothieware

```
Build version: ~11myvar2, Build date: Mar 20 2020 20:10:20, MCU: LPC1768, System Clock: 100MHz

  CNC Build 4 axis
```

#### Hardware model you are using (Make, and version)

BigTreeTech SKR v1.3

#### Host and Operating System running the Host

Repetier Host
  
#### Link to running config and M503 results

M503 and console log: https://pastebin.com/ezpTffvS
Config: https://pastebin.com/nbAJTb88

wolfmanjm, 2020-04-18:
Sorry we do not support BigTreeTech boards, I suggest you ask them for support.

closed 2020-04-18 by wolfmanjm

PKav89, 2020-04-18:
But it's clearly a firmware issue, not a board issue.
Also, i thought you don't like only MKS boards, BigTreeTech boards are open source.

wolfmanjm, 2020-04-18:
I don't think it is a firmware issue. on a real smoothie board P2.0 is on the alhpa driver, so the pins are complately different.
As for clone boards.. see here...  http://smoothieware.org/troubleshooting#what-is-wrong-with-mks

PKav89, 2020-04-18:
I tried to change pin to other pins, like 0.10 and 1.24, and issue still persisted.

There were only MKS boards in the blacklist before. I bought one by mistake and made a donation to the Smoothie project when I found it's not a real Smoothieboard. But i thought SKR boards are OK because they publish all schematics and documentation on Github. And it's even not a clone. The only same thing they use is LPC MCU.

wolfmanjm, 2020-04-18:
They use a counterfiet MCU you can see that in the version report.
As for pins I jave no idea what pins are being used on that board and how they are connected. It sounds like mauybe a pin is doubly defined or overloaded.

FWIW I have many machines here configured with kill buttons and 4 axis, and they all work fine. As kill module and homing module are totally separate I can't really see how one could affect the other unless there is a pin conflict.

PKav89, 2020-04-18:
There is no double definition in the configuration, I checked it.
Add kill_button lines to config - 4-th axis is dead. Remove kill_button lines - everything works.

The LPC1768 MCU is not counterfeit. It's just a little slower version of the same MCU, which does not seem to affect anything. And BigTreeTech recently created a new board with LPC1769, same as Smoothieboard. Will you support Smoothieware on those boards or only on the genuine Smoothieboard? 

Well, I'll try to debug it myself. If I find the solution, I will post it here.

wolfmanjm, 2020-04-18:
You should hook up an ftdi to the uart  port (if they have one) and see if the boot log reports any config errors. Sometimes a very long config file will run out of memory and some things do not get configured.

What I was also thinking was maybe the P2.0 is shorted to some other pin that affects the A axis motor.

wolfmanjm, 2020-04-18:
I just noticed the delta (a axis) enable is 2.12 which is the kill button on a smoothieboard. It maybe hard coded. (it  shouldn't be but it is always a kill buttopn on smoothieboards)

wolfmanjm, 2020-04-18:
yea that is the issue it is hard coded to be pause button...
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/utils/killbutton/KillButton.cpp#L41

This is one reason we don't really support other boards, no way to test them amongst other things.

PKav89, 2020-04-19:
But you always insisted on writing clean code and this board just allowed us to find an "unclean" line of code. I confirm,  it works correctly after A axis pins reassignment. 

So, maybe this line should be removed so kill button will be fully configurable? Or, otherwise, "kill_button_pin" config parameter should be removed, because i doubt somebody would want two kill buttons on a different pins?

wolfmanjm, 2020-04-19:
The code is the way it is to maintain backward compatibility with old config files for those that insist on NOT updating their configs when they update the firmware. At some point it will be removed. We do ty to maitain backward compatibility between releases. As I said ALL smoothieboads use that pin for the kill (and used to be pause) button and most clones do too. There is no reason IMO to consider boards that do not follow pinning practices, they should provide their own firmware.

wolfmanjm, 2020-04-19:
That pin is also hardwired in the bootloader to force it into DFU mode, so should NOT be used for other purposes unless yo also change the bootloader code and re flash the bootloader.

wolfmanjm, 2020-04-19:
FYI You could fix this issue in your config by defining `pause_button_pin nc`, although I'll be merging a minor fix which will ignore the `pause_button_pin` but will set `kill_button_pin` to 2.12 by default if not defined, which in your case will fix the issue as well as you define it to be different. Overall it is a bad idea to use pin 2.12 due to its use as the ISP pin in the bootloader.

wolfmanjm, 2020-04-19:
I also added documentation to the wiki about other pins which are hard coded and should be used with caution.

PKav89, 2020-04-19:
Yes, I confirm that `pause_button_pin nc` also works. And information about hardcoded pins in wiki is also very useful. Thanks for help!

Angeliqueratna, 2022-02-16:
Can u help me for our last school project🥲, i make cnc router 4axis with axis A is a rotary axis with spindle using Bigtreetech skr v1.3 and reprap lcd can u give me a example of the firmware thats support 4 axis with lcd reprap???

Angeliqueratna, 2022-03-10:
How to add axis A with mcs and wcs, jog A that like axis x y z in lcd reprap?