Smoothieware/Smoothieware #1459: M23 command with "m" or "g" letter makes file.open failed

lukrow80, 2020-04-23:
When using M23 command with m or g leter ther is a problem with parsing the command. The name of the file is cutted out to the m or g letter. Looking down in sourcecode there is a gcode->getcommand and there is option for is_m or is_g maybe this makes this problem.

I am using compiled verion of smothieware with CNC=1 and AXIS=5 PAXIS=5

wolfmanjm, 2020-04-23:
what do you expect to happen? this report is meaningless i am afraid, I have no idea what you are reporting.
M23 should not be used in a CNC setting, try using play file command instead. FWIW M23 worka as reprap specifies it should.

wolfmanjm, 2020-04-23:
Try making the file name lowercase, uppercase filenames would cause issues.

closed 2020-04-23 by wolfmanjm

wolfmanjm, 2020-04-23:
Ok I understand what you are saying now. (took a while). Basically M23 is a non compliant gcode and cannot be handled properly. However so long as the filename is lowercase it will work. Uppercase filenames are not supported for the reason you have discovered as M and G are control words and lines are split up on them. m and g would be ok.

In a CNC setting you should use the play command which will work (as it is all lowercase).