Smoothieware/Smoothieware #1460: G2/G3 as first command does not work

vbolshakov, 2020-04-30:
Executing G2/G3 as first command (with or without previous homing) not work. Motors produce "beep" sound and not move. Looks like motors disabled - gamma_homing_direction home_to_max, and Z axis fall down.

Tested for firmware.bin and firmware-latest.bin from repository on cartesian and corexy using original Smoothieboard v1 and on SRK 1.3.

wolfmanjm, 2020-04-30:
you need to set a feedrate, try adding an F500 or something. BTW it wokrs perfectly well if you issue the correct commands.

closed 2020-04-30 by wolfmanjm

vbolshakov, 2020-04-30:
just chech with F. Also not work.

wolfmanjm, 2020-04-30:
Then you are doing something wrong. Maybe you are not fully aware of how to setup a G2/G3 command. I suggest you read a gcode primer on G2/G3.

This is not a help forum, it is for issues. If you want gcode help use the IRC channel. Thank you

Then

wolfmanjm, 2020-04-30:
FWIW I use G2/G3 for cnc operations all the time it works exactly as needed. But using it correctly is difficult.