Smoothieware/Smoothieware #1463: Endstops broken; -endstop allows mechanical crash; +endstop causes hard fault firmware crash

ean365, 2020-05-03:
Stock edge Smoothieware appears to have broken endstop operation.
I can recreate two different issues that may be related (see bold italics below for quick summary)...

### Issue 1:
(This works with any axis;  X-axis used here as example.)

Send `G0X500` and wait for it to complete.
Send `G0X100` and then quickly hold the -X endstop in and keep holding it in for the rest of this test.
Smoothieware displays:  `Limit switch -X was hit - reset or M999 required`
Send `M999`
Smoothieware displays:  `WARNING: After HALT you should HOME as position is currently unknown`
Send `G0X100`  <--Make sure you are still holding in the -X endstop
**_Smoothieware ignores the endstop and moves the axis until it mechanically crashes into the hard stop._**

```
ok
G0X500
ok
G0X100
Limit switch -X was hit - reset or M999 required
ok
G0X100
!!
M999
WARNING: After HALT you should HOME as position is currently unknown
ok
G0X100
ok
```

This is an easy mistake to make, and Smoothieware should instead refuse to move in the direction of the already depressed endstop, and should again display the "Limit switch -X was hit - reset or M999 required" message.

### Issue 2:
(This works with any axis;  Y-axis used here as example.)

Send `G0Y100` and wait for it to complete.
Send `G0Y500` and then quickly hold the +Y endstop in.
Smoothieware displays many lines of:  `Limit switch +Y was hit - reset or M999 required`
**_Then, Smoothieware firmware crashes with the following:_**
```
**Hard Fault**
  Status Register: 0x40000000
    Forced
**MPU Fault**
  Status Register: 0x92
    Fault Address: 0x100073f4
    Stacking Error w/ SP = 0x100073f8
    Data Access
$T0b0c:addeadde;0d:2c120600;0e:addeadde;0f:addeadde;#1e
```

```
ok
G0Y100
ok
G0Y500
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required
Limit switch +Y was hit - reset or M999 required

**Hard Fault**
  Status Register: 0x40000000
    Forced
**MPU Fault**
  Status Register: 0x10
    Stacking Error w/ SP = 0x100073e8
$T0b0c:addeadde;0d:2c120600;0e:addeadde;0f:addeadde;#1e
```

Unable to do simple backtrace, dump memory, or other debugging through MRI at this point because of hard fault.

I suspect that the hard fault is, at least partially, due to multiple recursive `call_event(ON_IDLE...` and/or `call_event(ON_HALT...` calls between the endstop, stepper, and robot modules overflowing the stack, which may be waiting on some (maybe 1ms) timed event, degenerate race, but I have been unable to pinpoint and confirm it, yet.

I am still working on trying to isolate/debug it further.  Please confirm that you or others can recreate these faults.

_____________**Environment info:** _______________

Config file: [config.txt](https://github.com/Smoothieware/Smoothieware/files/4571425/config.txt)

**M115**
`FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-2b4519b, X-FIRMWARE_BUILD_DATE:Apr  9 2020 15:24:27, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-GRBL_MODE:0, X-CNC:0, X-MSD:1`

**M503**
```
; No config override
;Steps per unit:
M92 X792.00000 Y792.00000 Z792.00000
;Acceleration mm/sec^2:
M204 S500.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.02500 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X125.00000 Y125.00000 Z125.00000 S-1.00000
;Max actuator feedrates in mm/sec:
M203.1 X125.00000 Y125.00000 Z125.00000
;E Steps per mm:
M92 E1674.4000 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
M207 S1.0000 F2100.0000 Z1.0000 Q3000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F1500.0000 P57988
;E acceleration mm/sec²:
M204 E500.0000 P57988
;E max feed rate mm/sec:
M203 E35.0000 P57988
;PID settings, i_max, max_pwm:
M301 S0 P326.6000 I36.0870 D739.0000 X255.0000 Y255
;Max temperature setting:
M143 S0 P150.0000
;PID settings, i_max, max_pwm:
M301 S1 P40.4000 I3.9790 D102.0000 X255.0000 Y255
;Max temperature setting:
M143 S1 P300.0000
;Home offset (mm):
M206 X0.00000 Y0.00000 Z0.00000
;Probe feedrates Slow/fast(K)/Return (mm/sec) max_z (mm) height (mm) dwell (s):
M670 S5.00 K100.00 R0.00 Z384.00 H5.00 D0.00
;Probe offsets:
M565 X35.00000 Y0.00000 Z2.00000
```

wolfmanjm, 2020-05-03:
Nope it all works as expected for me. Never seen this issue. In theory as long as the endstop is triggered it should continue to issue HALTS. Although you should always move off the endstop before doing the M999.

wolfmanjm, 2020-05-03:
The crash certainly looks like a stack overflow or out of memory issue. If you run gdb it will show you where it crashed and give ayou a backtrace, i ca n see from the dump it dropped into gdb.

wolfmanjm, 2020-05-03:
Looks to me like you are not running a genuine smoothieboard either. what board do you have?

ean365, 2020-05-03:
It's easy to make the mistake of not manually coming off the endstop prior to M999 (this is also not so easy on some machines).  If you make this mistake, an easy mistake to make, the first issue above will occur -- smoothie will ignore that the endstop remains pushed in and run right through it.  I believe it is reasonable to expect Smoothieware to always respect the limit switch.

I don't know why you would not be seeing the positive limit firmware crash.  I edited above description for recreating this.  It will stop the first time it hits the endstop, but then while +axis endstop is still held in, after sending M999, it the firmware crashes.  Can you or someone please confirm this.  I have a hard time believing it is just me, at the moment.

FWIW -- I am testing using a cartesian bot with the latest "edge" firmware binary downloaded directly from GITHUB (i.e., not compiling it myself).  I even went back to a version from last year and found same results.  I would be surprised if others are unable to recreate these issues.  (Unless there is something horribly wrong in my config.txt, but I don't see it.)

wolfmanjm, 2020-05-03:
I agree if this were reproducible it would be a serious issue, however I frequently run into this situation where I hit the limit and do $X then I try to jog again, and it simply HALTS again. As i said I cannot reproduce this at the moment.

wolfmanjm, 2020-05-03:
If you can get a backtrace it would help. even after a hard fault the backtrace is availbale, although there maybe some tweaking needed, look at the MRI github page I think he explains how to debug a hard fault.

alkabal, 2020-05-04:
Hi

This is not really related, you can remove my message if needed.

Regarding to the soft endstop, this is really good to allow a halt if you ask to move outside the soft endstop but imo this need to be separated if request is from Gcode move or from jog/pendant move, if this is from jog do not halt only inform user or only do not move. (linuxcnc work like this, machine don't move but is not killed if this is from jog/pendant)

Br

ean365, 2020-05-07:
I have identified and located at least a couple of serious bugs.  This is definitely a problem to "obeying the limits".  The Endstopd module has seemingly had a bug from its beginning, and it was possibly made worse (hard fail due to stack overflow) over time by changes to the Conveyor module.  It is also possible (I think) that not everyone will experience the hard fault, depending on where in the ON_HALT and ON_IDLE event queues the Endstops module is relative to the Conveyor and StepperMotor modules (possibly due to config.txt file layout).  This should all become apparent as I explain...

The root of the problem lies in the `Endstops::on_idle()` code:
```
// only called if limits are enabled
void Endstops::on_idle(void *argument)
{
    if(this->status == LIMIT_TRIGGERED) {
        // if we were in limit triggered see if it has been cleared
        for(auto& i : endstops) {
            if(i->limit_enable) {
                if(i->pin.get()) {
                    // still triggered, so exit
                    i->debounce = 0;
                    return;
                }

                if(i->debounce++ > debounce_count) { // can use less as it calls on_idle in between
                    // clear the state
                    this->status = NOT_HOMING;
                }
            }
        }
        return;

    } else if(this->status != NOT_HOMING) {
        // don't check while homing
        return;
    }

    for(auto& i : endstops) {
        if(i->limit_enable && STEPPER[i->axis_index]->is_moving()) {
            // check min and max endstops
            if(debounced_get(&i->pin)) {
                // endstop triggered
                if(!THEKERNEL->is_grbl_mode()) {
                    THEKERNEL->streams->printf("Limit switch %c%c was hit - reset or M999 required\n", STEPPER[i->axis_index]->which_direction() ? '-' : '+', i->axis);
                }else{
                    THEKERNEL->streams->printf("ALARM: Hard limit %c%c\n", STEPPER[i->axis_index]->which_direction() ? '-' : '+', i->axis);
                }
                this->status = LIMIT_TRIGGERED;
                i->debounce= 0;
                // disables heaters and motors, ignores incoming Gcode and flushes block queue
                THEKERNEL->call_event(ON_HALT, nullptr);
                return;
            }
        }
    }
}
```

...which does a `call_event(ON_HALT)` when a limit switch is hit.  Then `Conveyor::on_halt()` gets called as part of the ON_HALT event queue, which calls a couple of other helper functions within Conveyor, included in snippet below:
```
void Conveyor::on_halt(void* argument)
{
    if(argument == nullptr) {
        flush_queue();
    }
}

/*
    In most cases this will not totally flush the queue, as when streaming
    gcode there is one stalled waiting for space in the queue, in
    queue_head_block() so after this flush, once main_loop runs again one more
    gcode gets stuck in the queue, this is bad. Current work around is to call
    this when the queue in not full and streaming has stopped
*/
void Conveyor::flush_queue()
{
    allow_fetch = false;
[... 45 lines cut ...]
```
                if(i->debounce++ > debounce_count) { // can use less as it calls on_idle in between
                    // clear the state
                    this->status = NOT_HOMING;
                }
 ```
Except for when the -X limit switch is hit, these lines will be run any time any limit switch is hit and remains held in for more than a few milliseconds.  This means that `i->debounce ` is continuously incremented (and rolls over) continuously for all axes "before" the one axis that has hit its limit switch. So, for instance, even if the -Y axis limit is held in, this code will run for the -X and +X endstops, and eventually, quickly, `this->status` will change to "NOT_HOMING", which essentially means "NOT_LIMIT_TRIGGERED".

When that happens, not "LIMIT_TRIGGERED", the next time `Endstops::on_idle()` is called, it will again fall through to run the second part of the on_idle code that again calls `call_event(ON_HALT)`.

This might be OK, but then ON_HALT event causes the `Conveyor::on_halt()` code to be called, and if you follow that code, you will see that it calls and loops on `call_event(ON_IDLE)`.  Now you should see where this is leading...

If Converyor is looping calling ON_IDLE event, and waiting for the machine to idle, and Conveyor is in ON_HALT queue before StepperModule, then StepperModule::on_halt() is never reached so that it can set `moving= false;`, so that the `is_moving()` check in `Endstops::on_idle` can break this infinite recursive event loop -- `Endstops::on_idle()` calls ON_HALT event, then `Conveyor::on_halt()` calls ON_IDLE event, which recursively calls `Endstops::on_idle()` until the stack overflows.

In fact, the best way to trace and reveal these bugs is to put a breakpoint on the 
`this->status = NOT_HOMING;`
line, trigger and hold-in a limit switch, breakpoint hit, and do a stack backtrace, and watch the stack grow steadily every few times around the "loop".

________
I have a few ideas on how to (and volunteer to) fix `Endstops::on_idle()` appropriately and also a way to improve event handling to make it much less likely that such "traps" can occur from events called from within events.

Calling events from within events, and especially looping on calling an event, seems pretty dangerous and fraught with potential pitfalls as the code base grows and gets more complex.  

[... cut ...] call that it made, it will use the static state variables to finish calling all the rest of the events in the particular queue it was running, before then starting and finishing the newly queued event.  This way no recursive event call could prevent another on_event from being called in proper turn.  When no more events are called within events, essentially when it would have normally returned to main, `Kernel::call_event()` will simply return to main, having run all event queues in proper turn.

wolfmanjm, 2020-05-07:
I'll need to look at this more carefully. I am not in favor of rewriting the event loop as that is overkill IMO. The easiest and correct fix is to have the hard limits checked in a timer (interrupt level)  rather than in on_idle, which should have been done in the first place. This will fix any issues with recursion in on_idle and will make the limit detection more sensitive. Looks like the debounce would need to be handled correctly too, but as the homing is handled this way I am not sure why the limits are different.

wolfmanjm, 2020-05-07:
The entire debounce in on_idle is also buggy as you point out. That is a good catch :) However by moving it into the endstop timer, the debounce can be the same as the endstops, and only checked for a limit that was trigered (that is also a bug at the moment, not sure why all of them are checked).
I do not believe the event handling is buggy at all, the main issue here is using on_idle for hard limit checking (that was done I think to avoid using more CPU time in the timer interrupt). on_idle everywhere else is pretty mundane and not likely to cause a stack overflow. Although I think commenting all on_idle loops to remind developers to not do too much in there. The main purpose of calling on_idle in the safe_wait is to avoid the watch dog timer going off, and to still deliver status messages immediatley when the query command (?) is recieved.
By not using on_idle in endstops the potential for recursion and unexpected issues will be removed. (unless you can find another place where on_idle is being misused).
So this is a good analysis, it has uncovered at least two bugs in the hard stop handling, but the conclusion that the event handling is flawed is incorrect IMO.

wolfmanjm, 2020-05-07:
I am still not sure why I cannot recreate this issue, as you say it may be the order in which modules are called.

wolfmanjm, 2020-05-07:
Ok calling flush_queue (which calls on_idle) from an on_halt which is also called from on_idle (as it cannot be called in an ISR) is also bad, I'll fix that too (somehow).

wolfmanjm, 2020-05-07:
Also found several modules calling things that may call on_idle when on_halt is called. on_halt was always meant to be safe to call from any context (except ISR). So I'll fix those edge cases too where I can.

wolfmanjm, 2020-05-07:
Here is my first attempt to fix all these issues... https://github.com/Smoothieware/Smoothieware/tree/fix/hard-limits

wolfmanjm, 2020-05-07:
Actually while working on this code again, I think I remembered that if you clear HALT caused by a limit switch hit, it deliberately lets you move the axis, presuming you know what you are doing, so you can jog the axis off the endstop, otherwise on some machines how would you be able to do that? It was a trade off, it was to retrigger halt then you would only be able to manually move the axis, and on some CNC machines you can't easily do that. So there is an issue here with fixing this as you would like it fixed. (the bugs aside).

wolfmanjm, 2020-05-07:
My initial way around this is to issue a warning.. "_WARNING limits are disabled until all have been cleared_". So continue with caution after endstops have been hit.
I know this is not ideal but at present I see no way to only limit movement away from a triggered limit switch. It would require the underlying motion control to be aware of limit switches which due to the module architecture is not easily done.
I can however add a config option to disable all movement until limit switches are manually cleared. For 3D printers this would be fairly easy for the operator to do, but they would need to do it manually as you would not be able to jog.
A current alternative is to enable soft endstops as well and select the config option to ignore movements that violates the soft endstop, this has the same effect as only allowing movement away from the limit switch. The only change I need to make then is to effectively set the homing when the endstop is triggered so S/W endstops know where they are and are enabled.

ean365, 2020-05-09:
FWIW, I did not mean to imply that event handling is buggy.  Certainly, `Kernel::call_event()` is pretty simple and tight, no bugs there, I agree.  I was asserting that, these bugs that I identified, which have existed for a long time, point out that it is easy to both misuse and misjudge consequences of using the event system -- especially as the code base grows, with multiple developers, and so many users, there are plenty of opportunities to fall into similar traps...
I've been thinking, and what follows is a discussion -- up for discussion:

1. `Kernell::call_event()` (events in general) is essentially the engine that runs Smoothieware.

2  Modules' `on_event()` code are called-back though `Kernel::call_event()` in the order in which they registered for a particular event in the first place.

3. The order in which modules register for event call-backs is to a significant extent governed by where they appear (maybe were enabled) in the `config.txt` file.

4. The order of where different modules appear in `config.txt` is, currently, by design, free-form, more or less at the whims of the individual user of Smoothieware.  This is generally good and desirable, but...

5. Back inside the firmware, it turns out that it can make a significant difference to the stability of Smoothieware and how Smoothieware performs and reacts to certain events contingent upon where in the event call-back order a particular module resides -- as #1463 has shown.

6. Smoothieware's events system and the way it is used by the modules, allows for one module's on_event call-back to (unwittingly) preempt, circumvent, and even prevent other module's on_event calls from running, if they appear later in the event call-back queue.

7. Given the above, with regard to the events system, as the code base grows more complex, it can be quite difficult to both
      a) predict the consequences of (recursively) calling `Kernel::on_event()` from within any module's `on_event()` call-back for any event.
      b) test the firmware sufficiently to be confident that it works regardless of the order in which modules registered for an event, and the order in which they are subsequently called-back.

So, how do we make things better, if we can, if we choose?  What are the options?

1. Comment all call_event and on_event code, and update the developer instructions, to be keenly aware of and avoid all of the pitfalls with regard to working with events.  <-- maybe the least we can do.

2. Forbid calling ANY event from within ANY other event.  (Probably antithetical/unworkable, but just mentioning to be thought-complete.)

3. Constrain developers to keep all `on_event()` code not only brief, but keep it all within that single call-back subroutine.  This would at least make it easier to see what other events are potentially called recursively.  However, this is also probably unworkable, not flexible enough.

4. Forbid calling ANY event from within the SAME event; e.g., `main()` calls 'call_event(ON_IDLE)`, which calls-back a module that again calls `call_event(ON_HALT)`, which calls-back a module that again calls 'call_event(ON_IDLE)` <-- ON_IDLE event has now been called from within itself -- forbidden.

5. Forbid any module, except for `main()`, from calling the ON_MAIN_LOOP and ON_IDLE events, and rewrite any existing modules that do.

6. Forbid any module's `on_event(ON_HALT)`, and anything called from within that, from calling ANY other event

7. Instrument `Kernel::call_event()` with some extra health monitoring code that, for instance, maintains a simple static count of how many times it has been called recursively, and either outputs a warning message, or even halts safely/preemtively, if the counter exceeds some preset threshold.  Also, keep a running tally of the maximum number of recursions, which can be interrogated by a command while running, or after running a long job, and deliberate testing of endstops and overtemps, etc., to give a rough indication of the "tightness and reliability" of the firmware -- is the firmware relying on too many levels of recursion and approaching the margins.  This maybe, also, the best (and only) place to simply monitor the stack for a near-overflow condition, and again ensure that the machine is safely/preemptively halted prior, or at least a warning message given.  Are some use cases running too close to the edge?  (Maybe, to some degree, the watchdog handles this, but by then it's too late, the fallback of last resort, something really bad happened.)  In any case, this would allow the entire user base to report potential issues that were otherwise difficult to test for, because of the variability of config.txt files, different machine configurations, etc.

8. I started by agreeing that `Kernel::call_event()` itself is not buggy, in fact quite simple and tight, which is good, but then again by nature totally permissive.  Maybe it can be made less naive about how it runs its event queues.  Rewrite the `Kernel::call_event()` code such that if/when it gets recursively called-back from within an `on_event()` call that it made, it will finish calling all the rest of the events in the particular queue it had been running, before then starting and finishing the newly queued event, and also skipping over recursively calling-back any particular on_event() that was already called (i.e., one that called `call_event()` recursively earlier in the "stack-up"). This way, at least, no recursive event call could preempt [... cut ...]
Although none of the above will solve all evils and prevent all pitfalls, nonetheless some combination may be better than nothing, and probably helpful to either alleviate some issues by making them less likely to be a problem, or by making it easier to identify potential problems and marginal/edge cases when or even before they occur.

Food for thought.....

(Let me know if perhaps this thread has moved or will move philosophically beyond this particular bug report, #1463, and is better continued in the developer forums???)

ean365, 2020-05-09:
Strictly with regard to Endstop handling....
I do suggest and would like to see a configuration option added that would allow a choice as to whether the machine or a particular axis could be moved at all when a limit switch is held in.
Or, maybe even options to allow Alkabal's comment above.
(Better yet, only allow movement away from the endstop -- but I understand the current difficulties with this and machines with common/parallelled endstops, though again maybe a config option?)

wolfmanjm, 2020-05-09:
if you don't want it to move then don't clear the halt. By doing M999 or $X you are explicitly stating "yes I know I hit an endstop, but I want to move anyway".

wolfmanjm, 2020-05-09:
Personally I don't think it worth the programming effort, but it would be relatively easy to add an option that would just retrigger ALARM until the limit is physically released, but as I said you won't be able to jog to release it which on some machines is an issue.

wolfmanjm, 2020-05-09:
I agree that in a perfect world where speed, ram and flash space was unlimited, and all programers are untrustworthy, it would be good to make the event system fool proof.
However I rejected the idea that the code needs to make up for programmer inabilities. This is embedded code, and trade offs are always required to shoe horn the functionality into limited resources. One trade off here is to hope that programmers know what they are doing, or the reviewers will know what they are doing (both failed in this case :) But for the most part we review PRs quite carefully, and we test as best we can. Edge cases will always slip though, as this one did, and it was a real edge case as no one in several years has come across it.
So I for one vote against any complications added to the event system.
FWIW on-idle is an odd case really and technically shouldn't even be an event. In V2 this problem does not even exist as there is no on_idle, as we run under an RTOS, which does do some basic checking. However even then FreeRTOS does not disallow someone putting an infinite loop in a Timer callback, they just document that you should not do that ;) and that is my philosophy here.We document that you must be careful to not take up too much time in on_idle or any other event callback. In fact if you do the watchdog will fire. We continue to review PRs very carefully, and add this to the list of things to check for.

wolfmanjm, 2020-05-09:
I am happy to continue the discussion, but there is a PR on the table to at least fix the crash, so I'll close this issue in favor of PR #1464

closed 2020-05-09 by wolfmanjm