Smoothieware/Smoothieware #1468: Possible issue with PID control on powerful bed heater

tschiex, 2020-05-24:
#### What were you doing?

I have changed my 12V bed heater for a powerful SSR controlled 220V bed heater that heats my bed very quickly. I autotuned the PID.

#### What did you expect to happen?

I expected it would still work fine :-).

#### What happened instead?

The PID controller seems unable to control the bed temperature which slowly rises as time goes. Then Smoothieware decides to HALT (Temperature too slow...."). The issue seems to come from the "random" temperature controller spikes that exist, even when the bed is too hot (get temp bed output):

[12:04:24:935] bed temp: 57.049042/50.000000 @0
[12:04:26:029] bed temp: 57.042342/50.000000 @0
[12:04:28:149] bed temp: 57.040590/50.000000 @0
[12:04:30:297] bed temp: 57.039376/50.000000 @73
[12:04:32:149] bed temp: 57.039001/50.000000 @0
....

The PID controller seems to correctly tune the output to zero but from time to time, the heater PWM is turned on (at 73/255 here). The heater being powerful, these small pikes are sufficient to continue heating the bed beyond the target temperature (especially when the weather is hot and the  bed does not loose "enough" heat).

Bang Bang does the job perfectly however (and that's what I'm using) so it's not a crucial issue at all (BTW, it could be nice to add a second parameter to bangbang: an offset such that 

if temperature  > target + hysteresis - offset    STOP
if temperature < target  - hysteresis - offset     ACTIVATE

Without this, my average heating temperature is always above target because of thermal inertia and the large heater power)

So, there may be a flaw in the actual PID controller code since these random spikes don't seem to come from noisy measures. I don't know why they are there but if it's on purpose, they should probably be turned off when the PID controller output is negative. I will try to find where this is added in the code.

#### Bra@nch & Commit or Version of Smoothieware

Build version: edge-2b4519b, Build date: Apr  9 2020 15:24:27, MCU: LPC1769, System Clock: 120MHz

#### Hardware model you are using (Make, and version)

    Azteeg X5 Mini V1 (I think)

#### Host and Operating System running the Host

Linux + Serial terminal (CuteCom)
    
#### Link to running config and M503 results

M115

[11:54:32:427] FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-2b4519b, X-FIRMWARE_BUILD_DATE:Apr  9 2020 15:24:27, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-GRBL_MODE:0, X-CNC:0, X-MSD:1␊

M503

[11:55:35:844] ; config override present: /sd/config-override␊
[11:55:35:844] ;Steps per unit:␊
[11:55:35:849] M92 X160.00000 Y160.00000 Z160.00000 ␊
[11:55:35:849] ;Acceleration mm/sec^2:␊
[11:55:35:849] M204 S2000.00000 ␊
[11:55:35:849] ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:␊
[11:55:35:849] M205 X0.05000 Z-1.00000 S0.00000␊
[11:55:35:849] ;Max cartesian feedrates in mm/sec:␊
[11:55:35:849] M203 X500.00000 Y500.00000 Z500.00000 S-1.00000␊
[11:55:35:849] ;Max actuator feedrates in mm/sec:␊
[11:55:35:849] M203.1 X500.00000 Y500.00000 Z500.00000 ␊
[11:55:35:849] ;Optional arm solution specific settings:␊
[11:55:35:849] M665 L292.0000 R165.5625␊
[11:55:35:849] ;Digipot Motor currents:␊
[11:55:35:849] M907 X1.10000 Y1.10000 Z1.10000 A1.10000 ␊
[11:55:35:849] ;E Steps per mm:␊
[11:55:35:865] M92 E305.0000 P57988␊
[11:55:35:865] ;E Filament diameter:␊
[11:55:35:865] M200 D0.0000 P57988␊
[11:55:35:865] ;E retract length, feedrate:␊
[11:55:35:865] M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988␊
[11:55:35:865] ;E retract recover length, feedrate:␊
[11:55:35:865] M208 S0.0000 F480.0000 P57988␊
[11:55:35:865] ;E acceleration mm/sec<0xc2><0xb2>:␊
[11:55:35:865] M204 E500.0000 P57988␊
[11:55:35:865] ;E max feed rate mm/sec:␊
[11:55:35:865] M203 E150.0000 P57988␊
[11:55:35:865] ;PID settings, i_max, max_pwm:␊
[11:55:35:865] M301 S0 P315.2000 I23.3050 D1066.0000 X255.0000 Y255␊
[11:55:35:865] ;Max temperature setting:␊
[11:55:35:865] M143 S0 P300.0000␊
[11:55:35:865] ;PID settings, i_max, max_pwm:␊
[11:55:35:865] M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255␊
[11:55:35:865] ;Max temperature setting:␊
[11:55:35:865] M143 S1 P300.0000␊
[11:55:35:865] ;Home offset (mm):␊
[11:55:35:865] M206 X0.00000 Y0.00000 Z-14.40000 ␊
[11:55:35:865] ;Trim (mm):␊
[11:55:35:865] M666 X-0.376 Y0.000 Z-0.094␊
[11:55:35:865] ;Max Z␊
[11:55:35:865] M665 Z305.380␊
[11:55:35:865] ;Probe feedrates Slow/fast(K)/Return (mm/sec) max_z (mm) height (mm) dwell (s):␊
[11:55:35:865] M670 S5.00 K100.00 R0.00 Z200.00 H5.00 D0.00␊
[11:55:35:865] ;Probe offsets:␊
[11:55:35:865] M565 X0.00000 Y0.00000 Z0.00000␊

Config at: https://gist.github.com/tschiex/0d83e488f7346b2f5706dc5b2163e2eb

wolfmanjm, 2020-05-24:
You probably need to manually tune the PIS, as autotune only gets you so far, Ther is not likely to be a bug in the PID as thius was taken directly from a widely used PID library that has been in use on thousands of platforms for many many years.

Manual tune is your answer (or continue using bang bang).
Also using PID on SSR is complex especially if it is an A/C SSR, as it needs to switch in sync with the AC which is not an option for us.

closed 2020-05-24 by wolfmanjm

triffid, 2020-05-24:
Your D term is far too strong.

You'll note that those power spikes happen when the temperature drops,
which is precisely what the D term does. It also cuts power while warming
up to prevent overshoots.

If you reduce D, you'll probably want to reduce P and I too otherwise
you'll get massive overshoot on setpoint step changes.

I've never had good results with PID autotune, it sometimes gets close-ish
to usable values but always requires tweaking afterwards.

I like to tune it so it gets 3-5°C overshoot during warmup, since that also
means that you hit the target temperature much faster since it's not
feathering onto the setpoint.

I don't think poor PID tuning is a code problem in Smoothie, although PID
is traditionally quite poor when the load response is highly asymmetrical -
ie it heats up significantly faster than it cools down.

triffid, 2020-05-24:
One of the reasons I wrote a ΣΔ modulator for smoothie instead of PID is
that it works significantly better with zero-crossing AC SSRs, just need to
set the frequency to either match line frequency or double line frequency,
ie 50/60Hz or 100/120Hz. Let the zero crossing detector handle sync.

Note that not all AC SSRs have zero crossing feature, but many do.

tschiex, 2020-05-24:
Thanks to all for all these quick and informative answers. 

I have accumulated some (theoretical and practical) knowledge of PID controllers as well as SSR and understand your points. My SSR is a Zero crossing one (not random) but that should not be the issue. It may not be very reactive but it will turn off every 10ms unless ordered to remain on. I'm just surprised it is asked to turn on when P and I should be so negative (especially I, after 20' of drifting above the target). Anyway, thanks again and sorry for bugging you. I will possibly do the tuning by hand then but bang -bang is fine really (I also changed the code to include the constant offset that suits me well :-).

Back to PPE printing :-)

tschiex, 2020-05-24:
I couldn't resist diving in the code and I think the issue is real, can effectively be minored by hand-tuning D  (making it smaller) but actually comes from a side-effect of the anti-windup strategy used in the PID controller (the 3rd line here, which is active whether this->windup is true or false, extracted from src/modules/tools/temperaturecontrol/TemperatureControl.cpp, lines 521 to 524)
```
    float new_I = this->iTerm + (error * this->i_factor);
    if (new_I > this->i_max) new_I = this->i_max;
    **else if (new_I < 0.0) new_I = 0.0;**
    if(!this->windup) this->iTerm= new_I;
```
Essentially, the new_I term, which is used to compute the command:

    this->o = (this->p_factor * error) + new_I - (this->d_factor * d);

can never be negative. Bounding the I term to prevent windup is very usual (there are other strategies) but always forbidding it from being negative looks extreme. This probably explains the behavior I see. 

There is an i_max (to prevent slow heaters from overshooting I assume) but the implicit i_min (equal to zero here) is too strong. It prevents I from cancelling "slow drifts above the target". A constant negative term would be better (and this is a usual choice I think). This code was likely written at a time when powerful heaters were infrequent.

Will do some tests when I find time and possibly report...

triffid, 2020-05-26:
The accumulator attached to the I term is supposed to find the static power
required to maintain temperature.

That's always more than zero if the set point is above ambient, and can't
sensibly be less than zero if the controlled device can't produce cold.

Only peltiers and phase-change systems can give negative power (ie generate
cold).

Regular heaters are incapable of generating negative power, therefore
cutting the accumulator off below zero is perfectly sensible.

Breaking how PID works with heaters because autoPID gave you bad numbers
isn't the answer.

Arguably what you're chasing here is improvements to the autoPID algorithm
- which I'm sure could be accepted if you produce good documentation, a
strong rationale, and clean code.