Smoothieware/Smoothieware #1472: firmware edge-dc4dcbe: some screens on reprap_discount_glcd  aren't updated

az143, 2020-06-03:
with this 10 day old edge firmware some screens no longer update correctly:
1. on the prepare/set temperature/bed and hotend screens rotation of the encoder doesn't cause the screen to update, so changes are blind. only after clicking the encoder do i see the newly/blindly set temperature on the parent screen.
(i didn't check any further configuration screens, but i suspect they are also now blind wrt. changes.)
2.  the probe/status screen doesn't show anything until after the encoder is rotated, and the switch/probe status is only updated when the encoder is rotated, not live as in the past.
3. interestingly, rotating the encoder on the watch page does update the speed factor as expected.

this has worked with the same, bog-standard panel config for all firmwares between
edge-9462444 from jan 2018 and up to & including firmware-edge-bdb141f from jan 2020.

here is my panel-related config snippet:
---
panel.enable true
panel.lcd reprap_discount_glcd
panel.spi_channel 0
panel.spi_cs_pin 0.16
panel.encoder_a_pin 3.25!^
panel.encoder_b_pin 3.26!^
panel.click_button_pin 1.30!^
panel.buzz_pin 1.31
panel.back_button_pin nc
panel.menu_offset 0
panel.encoder_resolution 4 # less jumpy please

panel.alpha_jog_feedrate 4500
panel.beta_jog_feedrate 4500
panel.gamma_jog_feedrate 360

panel.hotend_temperature 185
panel.bed_temperature 60
---

wolfmanjm, 2020-06-03:
Please try PR #1473 and let me know if it fixes the issue. Thanks

wolfmanjm, 2020-06-03:
Ok I merged PR 1473, which reverts the update that broke this. I'll update the firmware-latest.bin with this latest build. Thank you

wolfmanjm, 2020-06-03:
If anyone wants to fix the PR #1413 That broke this please let me know.

closed 2020-06-03 by wolfmanjm