Smoothieware/Smoothieware #1481: Crash on Bigtreetech SKR 1.4 + TFT 3.5 V3.0

DrTon, 2020-08-05:
#### What were you doing?

Printing big models

#### What did you expect to happen?

Printer prints

#### What happened instead?

On big models printer halts after random period (around 30 min to 3 hours). If I increase speed to 300% (now testing without filament) it halts more early (10-30min).

#### Branch & Commit or Version of Smoothieware

```
Build version: edge-bdb141f, Build date: Jan 12 2020 10:21:20, MCU: LPC1769, System Clock: 120MHz
5 axis
```

M115 output:
```
FIRMWARE_NAME:Smoothieware, FIRMWARE_URL:http%3A//smoothieware.org, X-SOURCE_CODE_URL:https://github.com/Smoothieware/Smoothieware, FIRMWARE_VERSION:edge-bdb141f, X-FIRMWARE_BUILD_DATE:Jan 12 2020 10:21:20, X-SYSTEM_CLOCK:120MHz, X-AXES:5, X-GRBL_MODE:0, X-CNC:0, X-MSD:1
```

#### Hardware model you are using (Make, and version)

- Bigtreetech SKR 1.4 Turbo
- TFT 3.5 V3.0 connected via UART and works as gcode sender

#### Host and Operating System running the Host

Linux x64

#### Link to and console logs

Truncated thousands of ok, they don't fit in console buffer.

```
                                              ok
                                                ok
                                                  ok
$T050c:0065e500;0d:207f0010;0e:c3b50000;0f:c2b50000;#3d
```
    
#### Link to running config and M503 results

M503 output:
```
; No config override
;Steps per unit:
M92 X160.00000 Y160.00000 Z800.00000 
;Acceleration mm/sec^2:
M204 S2000.00000 Z200.00000 
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X100.00000 Y100.00000 Z33.33333 S-1.00000
;Max actuator feedrates in mm/sec:
M203.1 X100.00000 Y100.00000 Z33.33333 
;Digipot Motor currents:
M907 X0.80000 Y0.80000 Z0.80000 A0.80000 
;E Steps per mm:
M92 E1674.0000 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F480.0000 P57988
;E acceleration mm/sec²:
M204 E500.0000 P57988
;E max feed rate mm/sec:
M203 E50.0000 P57988
;PID settings, i_max, max_pwm:
M301 S0 P51.0000 I3.9130 D166.0000 X255.0000 Y255
;Max temperature setting:
M143 S0 P300.0000
;PID settings, i_max, max_pwm:
M301 S1 P314.0000 I21.3000 D1160.0000 X255.0000 Y255
;Max temperature setting:
M143 S1 P300.0000
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00 
```

Config: https://gist.github.com/DrTon/4b3503d720545a691974834ae6d5c0ac

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

wolfmanjm, 2020-08-05:
We do not support Bigtree smoothie board clones, I suggest you get them to help you. Sorry.
We do not have anu of these clone boa5rds so we have no way to test or reproduce your issue.

wolfmanjm, 2020-08-05:
Printing over UART is also an issue due to serial line corruption. I suggest you try printing over USB and/or from the sdcard and see if you have the same issue.

If you really want to find out where it is crashing you need to to a crashdump and debug with MRI , as explained in the Wiki.
If you do provide the requirted crash dump I will look at it.

closed 2020-08-05 by wolfmanjm

arthurwolf, 2020-08-05:
This might be enlightning also: http://smoothieware.org/troubleshooting#what-is-wrong-with-mks

While most members of the project are unlikely to help you with SKR issues, if you first ask SKR for help, and they fail at providing it, you can try contacting me and will generally help at that point if I'm not too busy.

DrTon, 2020-08-05:
> Printing over UART is also an issue due to serial line corruption. I suggest you try printing over USB and/or from the sdcard and see if you have the same issue.

OK, thank you. Looks that I should forget about Smoothieware and use alternative firmware...