Smoothieware/Smoothieware #1498: M48 not documented in supported-g-codes section

mvadu, 2020-12-13:
Looking at [z-probe code, it does support M48](https://github.com/Smoothieware/Smoothieware/blob/c781f5dfc6e0e44f995c713cad48c0df729dfc55/src/modules/tools/zprobe/ZProbe.cpp#L362).

But both  [supported-g-codes section](http://smoothieware.org/supported-g-codes) on smoothie website, as well as [M48 documentation on RepRap ](https://reprap.org/wiki/G-code#M48:_Measure_Z-Probe_repeatability) seem to indicate M48 is not supported in Smoothie. Can we please update this important G code?

wolfmanjm, 2020-12-13:
They are public wiki's please add it yourself.

closed 2020-12-13 by wolfmanjm

arthurwolf, 2020-12-13:
Hey there. 

Documentation is always going to lag a bit behind code it's mostly unavoidable.
If you want to add it yourself don't hesitate to. If you don't want to, tell me and I'll add it to my todo list.

Cheers.

mvadu, 2020-12-13:
@wolfmanjm are there any instructions on how to edit the documentations? Clicking on page source states to contact admins, which is exactly I did.
[image]

@arthurwolf I will be glad to do it if there are any "How To" somewhere.

arthurwolf, 2020-12-13:
Do you have an account on the wiki, or are you trying to anonymously edit
it?

wolfmanjm, 2020-12-13:
you can edit any page except the main page. you need to create an account and login.

mvadu, 2020-12-13:
Ah, I hadn't noticed that login option there. I have created an account now.  I think Account creation might be broken, it complained about a DB connection, but I did get the email.
[image]

mvadu, 2020-12-13:
I was able to edit the page, thank you @wolfmanjm @arthurwolf !

arthurwolf, 2020-12-13:
Thanks for improving the documentation.