Smoothieware/Smoothieware #1501: Commit edd99f33fe06cfdf3cf1d74e28f921366f59a56b broke Max31855 and maybe other sensors

sendel, 2021-02-19:
It read temperature by Idle, but it not good state for read, and temperature can not change long time. And it add on_idle to Max31855.cpp/.h but other sensors havent it function.

wolfmanjm, 2021-02-19:
on_idle() is the only place SPI can be read, it runs about every 100ms so is perfectly adequate for temperature readings.
I suggest that your problem is that you either do not have a functioning SPI or are using the wrong SPI port.
I do not think it has anything to do with updates happening in on_idle().

closed 2021-02-19 by wolfmanjm

arthurwolf, 2021-02-20:
Can you describe more precisely what you mean by "havent it function"? What exactly are you seeing that is an issue?

sendel, 2021-02-22:
> Can you describe more precisely what you mean by "havent it function"? What exactly are you seeing that is an issue?

Not shure, but on_idle function Im found only for Max31855, and not know what will be when idle process call sensor.on_idle for other sensor, Thermistor for example. And other "problem" Max31855 havent function/params that have other sensors, void get_raw(); and float min_temp, max_temp;

arthurwolf, 2021-02-22:
What do you mean you're not sure? If something is not working correctly, you should be able to tell us what is not working. Are you sure anything is in fact not working ??