Smoothieware/Smoothieware #1504: Memory leak handling M115?

SyntaxColoring, 2021-03-04:
I haven't had a chance to confirm this by running it through Valgrind or anything, but reading the code, I *think* memory is leaked every time the device handles a `M115` command.

A `Gcode` object is allocated with `new` here...

https://github.com/Smoothieware/Smoothieware/blob/5857002a87f66ff67daa888f72f17a9eb78006a9/src/modules/communication/GcodeDispatch.cpp#L155-L156

And then never `delete`'d here...

https://github.com/Smoothieware/Smoothieware/blob/5857002a87f66ff67daa888f72f17a9eb78006a9/src/modules/communication/GcodeDispatch.cpp#L269-L291

A quick fix would be to `delete gcode` before the `return`, like the other `case` blocks do...

https://github.com/Smoothieware/Smoothieware/blob/5857002a87f66ff67daa888f72f17a9eb78006a9/src/modules/communication/GcodeDispatch.cpp#L261-L267

But the better fix, IMO, would be to make this bug impossible. Why is `gcode` allocated with `new` in the first place? Can we refactor it so it's placed on the stack, instead?

arthurwolf, 2021-03-04:
It does look like this command does forget to delete the Gcode yes, looking
at all the other return instances.

arthurwolf, 2021-03-04:
Looks like this might have been forgotten when the command was first added,
then never noticed since:
https://github.com/Smoothieware/Smoothieware/commit/620b6dffe8b1c646aad09d425991666a058c0ec3

wolfmanjm, 2021-03-04:
good catch. fixed in commit 36ee3c61

closed 2021-03-04 by wolfmanjm

wolfmanjm, 2021-03-04:
FYI it can;t go on the stack as stack space is extremely limited, and this can be called recursively.