Smoothieware/Smoothieware #1505: Feature Request: Show temperatures of Temp module on CNC Screen

dreiduke, 2021-04-22:
Smoothieware
CNC built
Temperature Module
Temperature Switch module
 with RRDGD   Panel

#### What did you expect to happen?
Want to show Temperature values of Thermistors, for example water temperature or PSU temperature
and/or graphical alerts, if a certain temp is reached or overshoot
on the attached graphics Display (reprap discount or compatible)

#### What happened instead?
nothing

#### Branch & Commit or Version of Smoothieware
last  edge,  with temp modules activated, AXIS=4, CNC=1, 

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Copy-paste this template **completely**. Do not skip any lines!

It might happen that you are asked to provide a more thorough crash log which you can find out how to do [here](http://smoothieware.org/mri-debugging
)

wolfmanjm, 2021-04-22:
This would be a nice idea, but thermistors are disabled in the CNC build. You have to have a special build to enable them.
There is also very limited screen real estate in the CNC mode to display anymore info.
Having said that a pull request for this feature would be considered if done carefully.

FWIW we are moving away from LCD screens towards graphical controllers like Smoopi where this sort of thing is much easier to do.

Thank you for the suggestion, but as the core developers are unlikely to implement this it will be closed.

closed 2021-04-22 by wolfmanjm

arthurwolf, 2021-04-22:
Hello dreiduke!

Thanks for the code!

Could you look into adding code that detects whether the temperature
control stuff is compiled in, and *only* if that code is built-compiled-in,
then the screen is changed to have the extra information, and otherwise,
the screen stays the same ?

Cheers!

dreiduke, 2021-04-22:
I Compiled with this modified options in the Makefile
make AXIS=4 CNC=1 DISABLEMSD=1

This build works fine, also Temp readings via GCode works fine, 
also the console input "get temp" replyes with a temp value
So the only one, someone with Programming skills could make it visible on the panel

```
ifeq "$(EXCLUDE_MODULES)" ""

ifeq "$(CNC)" "1"
# CNC build excludes these    tools/temperatureswitch  tools/temperaturecontrol   <--- normally at the end of next row
export EXCLUDE_MODULES = tools/filamentdetector tools/scaracal tools/extruder 
else
# 3D build excludes these
export EXCLUDE_MODULES = tools/drillingcycles tools/spindle
endif

endif
endif
```