Smoothieware/Smoothieware #1524: G92 latency for closed loop offset/correction on the fly

Juanduino, 2022-08-28:
Dear all

Im doing a CNC build and in that process, im researching how a simple low cost closed loop “sub_driver” could be implemented. 

Have been going back and forth with the UGS crew about doing a virtual machine closed loop implementation as a UGS plugin, but there are several drawbacks. 

https://github.com/winder/Universal-G-Code-Sender/discussions/1967

Arguments against doing it would be:

Mainly Latency. 

Those hybrid-stepper add on things are very cheap. 

How would you implement a dedicated encoder/decoder sub_driver using the Arduino encoder lib by Paul@Teensy or the hardware QDEC on STM MCUs (which can possibly work w. Several encoders). 

One argument for streaming the sensed (physical) positions into UGS is the human/machine interface/learning/awareness part of it. I see some potential for comparing the physical position w. Gcode path in order to correct any offset by eg. G92 offset. That would involve some latency, but may still safe a print or CNC job from going bad, since it will correct lost steps before they possibly accumulate and the user would become aware of it.

Another option, I guess, would be to hook up the encoder “sub driver” to smoothie?

How fast does smoothieware implement G92 ? Can it be optimized?

Juanduino, 2022-08-29:
Nicely done!

So the analog value is part of the smoothieware planner or where exactly does it add or subtract the steps? Can it correct already planed moves by a step or two?

harlock999, 2022-08-29:
Smoothie is unaware of the analog value. Math and logic are done externally with the PLC, you can probably use an arduino derivative to do the same task. Depending on your implementation it can be very complex. In my case it corrects by a certain amount of steps every 10ms or so. Note that I barely use the Z movements so this is why my approach works.

If you plan to work in the Smoothie firmware code, be careful not to mess too much with the planner code. This is somewhat key to maintain a smooth toolpath.

Juanduino, 2022-08-29:
Ok, it’s just for Z axis. Still it’s nice.

I was thinking to have the combined position for X, Y (Y1&2) and Z being streamed into UGS, from a single MCU via USB packets holding the entire coordinate. The UGS plug-in should maybe just hold the Gcode path which it streams to smoothieware and discard what has already been passed (compared) that should make the program less memory heavy.

wolfmanjm, 2022-10-27:
Ad mentioned this needs to be done by an external processor and cannot be done within smoothie.

G92 is queued along with all other GCodes so cannot be made to work immediately.

closed 2022-10-27 by wolfmanjm