Smoothieware/Smoothieware #1526: lightburn pause problem

lateforwork1000, 2023-07-19:
Hay I'm having this problem with lightburn when I pause a job and unpause my laser no longer fires and I have to stop the job and reset just to get the laser working again. Is there a solution for this? 

this started to happen when i updated to the latest firmware.

wolfmanjm, 2023-07-19:
What command does lightburn send to pause smoothie? FWIW Smoothie does not actually support immediate pause other than stop sending gcodes (which is not immediate). There is suspend which is M600 which is also not immediate. We do not support the GRBL pause which I think is ~. Let me know what it is sending and I'll see why the laser may stop firing, at best we should ignore the command.

wolfmanjm, 2023-07-19:
Ok after reviewing the code, I do support ! to pause however unlike grbl it is NOT immediate. I suspect that lightburn is also sending an M5 when you pause (this is technically not required in smoothie as the laser will turn off when the block queue is empty). But M5 also sets the current sticky value of S to 0, and as lightburn does not set S on every G1 move the laser will effectively stay off until S is set again. S is set to zero because on a CNC S is the spindle speed and M5 needs to turn off the spindle, but on a laser M5 is used to turn off the laser. I am only guessing as I do not have lightburn to test with. However I did make a recent change that had M5 set S value to 0 for CNC machines using spindle control, and as this is a regression I will attempt to change the behavior so if the laser is enabled S is not set to 0. This may fix your issue (but again as I am guessing as to what is happening I cannot be sure). I'll close this issue when I checkin the changes.

wolfmanjm, 2023-07-19:
I have updated the latest firmware bins and potentially fixed this issue. If it does not fix it please report here.

closed 2023-07-19 by wolfmanjm

lateforwork1000, 2023-07-20:
thanks this fix the problem. I do have another issue that I found, I'm using lightburn and when I click the stop button my machine will stop but it will lock up telling me to use $X to unlock. the earlier versions did not have this problem. Another problem I'm looking into is steps per mm I it seems like the something is adding steps and or sub tracking steps. 

example if I calibrate my machine steps per mm to a 5inch square and then cut out the 5inch square it will be perfect, but when I cut out a 4inch square its off a little (3.98 of a inch) and if I cut out a 6inch square its is still off by a little ( 6.02) but if I cut out a 7inch square it goes higher (7.04) the same is when I cut out a 3inch but it sub tracks( 2.96 of a inch ) I do have a arduino controller and it works just fine no odd adding or subtracking steps. I'm still looking into this not sure why. 

thanks a lot.