Smoothieware/Smoothieware #1531: Issue reports for a non-connected printer so no command M115 output?

otede, 2023-12-09:
How would I go about reporting issues with Smoothieware config.txt when I haven't even succeeded in connecting my no-name printer to a terminal?

Smoothieware seems to regard my config.txt as corrupted (so the printer won't respond) if I dare to alter a single character in the serial communications section. I can't even alter baud rate for USB from the already set 250000. Changes in other parts of the config.txt do not cause such issues and do have an effect. I obey the encoding (ANSI) and end of line standard (Unix(LF)).

[config - works.txt](https://github.com/Smoothieware/Smoothieware/files/13625664/config.-.works.txt)
[config - doesnt work.txt](https://github.com/Smoothieware/Smoothieware/files/13625665/config.-.doesnt.work.txt)

Please don't trash the issue report. I'm willing to provide any details necessary.

Printer in question is an end-of-life startup project based on MKS SBASE 1.3 https://lumeplus.wixsite.com/lume
The firmware I have is from 2016/2017 and has been provided by the manufacturer of the printer.

wolfmanjm, 2023-12-09:
`uart0.baud_rate 250000`  is for the uart not usb. the usb does not have, nor need, a baudrate set.

I suggest you hook up an ftdi to the uart and see if you get any error messages.

I also suggest you disable as much of the config you do not need (like network and panel etc) as you may well be running out of memory, start with as little as you can enabled, then enable stuff as you need it.

However we do not support MKS SBASE products for reasons amply discussed elsewhere, so I would suggest contacting them for further support. We do not have any of those boards to test with.

Best of luck.

closed 2023-12-09 by wolfmanjm

otede, 2023-12-09:
I wouldn't know but even the config comment itself mentions the purpose of the rate setting is for connecting to pronterface/terminal which is what I'm trying to acomplish.
```
second_usb_serial_enable                     true            # This enables a second usb serial port (to have both pronterface
                                                              # and a terminal connected)
```
Also this line is one of the only two lines mentioning "usb", the other one pertaining to storage devices.

otede, 2023-12-09:
https://github.com/kliment/Printrun/issues/1393#issuecomment-1821365335

wolfmanjm, 2023-12-09:
you connect to pronterface over usb, not serial uart, so that baud rate setting is irrelevant and should be commented out

otede, 2023-12-10:
Thank you for bearing with me @wolfmanjm 

So I get that in case of such an error https://github.com/kliment/Printrun/issues/1393#issuecomment-1821365335
There's nothing Incan do in my end? It either connects or it doesn't? This was my last hope of connecting the printer since its Ethernet module is apparently dead. 

I might flash a new firmware one day but I'm not eager to do it with that unsupported trash (both the board and the whole printer).

otede, 2023-12-10:
Update:
I was able to connect via COM3@9600 (250000 baud rate value in the smoothieware config is still untouchable). Not only that, it seems that the whole MONITOR tab in Cura has gone live with all the functionality!

What I did: **Run as Administrator**

My very first G-Command issued:
>>> M503
SENDING:M503
; No config override
;Steps per unit:
M92 X200.00000 Y200.00000 Z3200.00000
;Acceleration mm/sec^2:
M204 S3500.00000 Z300.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X166.66667 Y166.66667 Z8.33333
;Max actuator feedrates in mm/sec:
M203.1 X300.00000 Y300.00000 Z8.33333
;WCS settings
G54
;Digipot Motor currents:
M907 X0.25000 Y0.25000 Z0.25000 E0.30000
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00
;E Steps per mm:
M92 E196.1200 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F480.0000 P57988
;E acceleration mm/sec²:
M204 E500.0000 P57988
;E max feed rate mm/sec:
M203 E50.0000 P57988
;PID settings:
M301 S0 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S0 P300.0000
;PID settings:
M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S1 P300.0000

**M115 still gives no return.**

wolfmanjm, 2023-12-10:
you have such an old firmware that it does not support M115. so do not worry about that.

otede, 2023-12-10:
Is it even worth upgrading? Does the new post-2017 firmware affect actual printing process? Someone told me smoothieware has undergone a significant upgrade sometime in 2019, with some config.text reorganisation. I'd gladly research it myself but the releases section is empty
https://github.com/Smoothieware/Smoothieware

wolfmanjm, 2023-12-10:
Yes there were significant upgrades, much smoother acceleration amongst other changes.

otede, 2023-12-10:
Again, thanks for all your help and for bearing with me despite my board being MKS SBASE.